nasa/cFE #1342: Clarify use of `Zero[4]` in `CFE_ES_CDS_AccessCacheData_t`

skliper, 2021-04-13:
**Is your feature request related to a problem? Please describe.**
Use of `Zero` not clear:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/modules/es/fsw/src/cfe_es_cds.h#L128

**Describe the solution you'd like**
Improve documentation/justify use.

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-04-16:
This union member is nothing more than an arbitrary block of raw data, it happens to be used when the CDS is cleared by filling it with zero bits and writing that to CDS, hence the name `Zero`.  

Propose to rename the field to `GenericDataBlock` to make it clearer that its just a bunch of bits.

skliper, 2021-04-21:
Closing to #1404, to improve how this works since the uint32 Zero[4] doesn't make much sense no matter what you call it in terms of interfacing with real persistent storage and how it's used in the CFE_ES_ClearCDS function.

closed 2021-04-21 by skliper