nasa/cFE #1345: Task not found case not explicitly handled in CFE_ES_RunExceptionScan

skliper, 2021-04-14:
**Is your feature request related to a problem? Please describe.**
Not clear if task not found case is handled correctly:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/modules/es/fsw/src/cfe_es_erlog.c#L360-L362

**Describe the solution you'd like**
Analyze and explicitly handle if needed or document why not needed for future reference

**Describe alternatives you've considered**
None

**Additional context**
Code Review

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-04-19:
Checked this, its fine as-is.

The default behavior after exception is to do a restart of some form  (processor or poweron).  Only if the exception is specific to a certain task __and__ the app owning that task was configured with "RESTART_APP" option, then only that app will be restarted instead of the whole system.

So in this case if the task was not found it will just fall through as intended to do the full processor restart.

Most likely if this occurs its because data structures are out of sync, which in an exception handling context is most likely because the data structures were possibly clobbered as part of whatever went wrong, before the RTOS actually caught the exception and stopped the task.  This can have far reaching effects/unpredictable behavior, so the best bet is to restart ASAP.

I suppose we could add a SysLog if that happens, but even that involves taking a lock which if the data structures are corrupted, could deadlock.  All hypothetical though....

skliper, 2021-04-19:
Recommend a comment to document this intended behavior.

PR #1385 (2021-04-28): Fix #1345, exception logic when app/task is not found
**Describe the contribution**
Adds comments to describe the logic when an exception cannot be traced back to a specific app, in that it should fall back
to the PSP reset.

Restarting only an app is a special opt-in case that is only done if specifically selected when starting the app, and the
exception is also positively traced back to that app.

Fixes #1345 

**Testing performed**
Build and sanity check CFE

**Expected behavior changes**
None - documentation only

**System(s) tested on**
Ubuntu 20.04

**Contributor Info - All information R [... cut]
commit d9f125bd (2021-04-28): Merge pull request #1385 from jphickey/fix-1345-app-not-found-doc

Fix #1345, exception logic when app/task is not found
commit e0a90796 (2021-04-28): Fix #1345, exception logic when app/task is not found

Adds comments to describe the logic when an exception cannot
be traced back to a specific app, in that it should fall back
to the PSP reset.

Restarting only an app is a special opt-in case that is only
done if specifically selected when starting the app, and the
exception is also positively traced back to that app.
closed 2021-04-29 by astrogeco