nasa/cFE #1360: Limit send HK MID's to 1 in pipe

skliper, 2021-04-14:
**Is your feature request related to a problem? Please describe.**
Default limit of 4 is excessive, getting behind is an error (shouldn't spam them either way).

ES example:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/modules/es/fsw/src/cfe_es_task.c#L405

**Describe the solution you'd like**
Set to 1

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2024-01-16:
@skliper you mean like this?
```c
Status = CFE_SB_SubscribeEx(CFE_SB_ValueToMsgId(CFE_ES_SEND_HK_MID), CFE_ES_Global.TaskData.CmdPipe, CFE_SB_DEFAULT_QOS, 1);
```

skliper, 2024-01-16:
Yes, that looks right.