nasa/cFE #1378: Documentation builds should use separate directories

jphickey, 2021-04-19:
**Is your feature request related to a problem? Please describe.**
The 3 documentation targets (detaildesign, usersguide, osalguide) all share the same "doc" subdirectory for the doxygen run.  This can become a problem when trying to validate documentation builds and building with the `-j` option.  They work fine when built one at a time, but with `-j` at one point my machine got into an endless loop with two doxygen builds running in parallel that kept stepping on the other's files.

**Describe the solution you'd like**
Use separate dirs so that builds can be done in parallel without interfering with each other.

**Additional context**
This should also result in separate "warnings.log" files too, that can be more easily checked.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2021-04-19:
Related to #389

jphickey, 2021-05-28:
After nasa/osal#1033 and #1561 this fixes half of the issue - now just between the detail design and user guide sharing the same subdirectory and "warnings.log" file.

PR #1612 (2021-06-15): Fix #1378, 1589, doxygen cleanup
**Describe the contribution**
Clean up and simplify the generated `Doxyfile`s for the documentation builds.  Instead of having special rules or providing direct refs to non-cFE subdirectories in the INPUT, just add the base dir for all modules.  Then use "exclude" logic to filter out parts based on file/directory names (unit-test, ut-stubs, etc).

Gives each cFE doc build (detaildesign, usersguide) its own dedicated working directory.

Fixes #1378 
Fixes #1589 

**Testing performed**
Build detail design and users guide docs and verify contents.

Built all 3 doc [... cut]
commit d89c7ce7 (2021-06-15): Fix #1378, simplify doxygen configs

Put the "detaildesign" and "users_guide" into separate working
directories, and do a general cleanup/simplification of the generated
doxyfiles.

Detail design just includes all components, and use "exclude" where
special treatment was needed, and to avoid UT code.

Paths for each "arch" (target-specific) are written to separate files
and included directly in the generated doxyfile, reducing the number
of generated files/layers.
commit e54e908e (2021-06-15): Merge pull request #1612 from jphickey/fix-1378-1589-doxygen-cleanup

Fix #1378, 1589, doxygen cleanup
closed 2021-06-16 by astrogeco