nasa/cFE #1381: CFE_ES_STOP_APP_CC results in processor reset (SP0 only)

dmknutsen, 2021-04-19:
**Describe the bug**
The CFE_ES_STOP_APP_CC is resulting in a processor reset when sent to our SP0.  Note that this behavior is isolated to the SP0 and is not observed on a Linux system.

**To Reproduce**
Start a test app
Send the CFE_ES_STOP_APP_CC command
Observe reset

**Expected behavior**
Application exits cleanly

**System observed on:**
SP0
OS: VxWorks 6.9
cFS Versions: cfe v6.8.0-rc1+dev218, osal v5.1.0-rc1+dev109, psp v1.5.0-rc124

**Reporter Info**
Dan Knutsen
NASA Goddard

klystron78, 2021-04-20:
Does it say anything about an exception occurring? Are you running the HS
app?

dmknutsen, 2021-04-20:
Nope - don't see an exception + we are running HS, but are using a test app that would not be included in any of HS's default table(s).  I power cycled the unit to ensure that HS's tables had not been modified by a previous test (to possibly include the app) and observed the same response.  Further, I attempted to stop a separate App (MD - also not included in the current version of HS's default tables) and observed the same processor reset response.  Attached is the log from the test run:
[cfs_stdout.txt](https://github.com/nasa/cFE/files/6343823/cfs_stdout.txt)

dmknutsen, 2021-04-20:
Also - I just removed HS from the startup script to ensure that it was not culprit.  The processor reset occurs following the stop App command without HS running.

jphickey, 2021-04-20:
Which app are you trying to stop?  Does this occur with _any_ app, or only a particular app?

Can this be reproduced with the basic cFE framework only (i.e. without any of the custom apps loaded, and with the default sample config)?

klystron78, 2021-04-20:
I'd really expect a printout from the PSP catching an exception. Does your
PSP print anything out on exceptions when resetting? What is the reset
status? Last I heard SP0 still clears all RAM so there's no CDS. (need a
modification to boot loader's DDR initialization code to prevent this)

Steven

dmknutsen, 2021-04-20:
Please see the attached file - In this case I sent a stop MM command (as opposed to a stop 'test_app' or stop MD in the previous test cases that result in a processor reset).  This case does not result in a reset - but we get a bunch of errors that should be useful in troubleshooting.
[cfs_stdout_stop_MM.txt](https://github.com/nasa/cFE/files/6344317/cfs_stdout_stop_MM.txt)

dmknutsen, 2021-04-20:
Worth note - I sent the CFE_ES_QUERY_ALL_TASKS_CC command prior to sending the stop app command - in the above logs task 0x10014 corresponds to MM and task 0x10004 corresponds to ES_BG_TASK.  I'm not sure how close I can get to the 'default sample config' when running against the SP0/JSC's FSW...I could try removing all/most of the apps from the startup script (with the exception of ci and to).  Do you think that would be worth trying?

klystron78, 2021-04-20:
I don't think it's worth trying. All the errors from OS_Lock_Global and
OS_Unlock_Global are telling. I think it's probably something with the
vxWorks OSAL that Joe Hickey can probably solve in 30 seconds of looking at
it. Otherwise I can try and find time to run vxWorks on my sp0 and
duplicate your problem.

You can see at the bottom it says "reboot()" so something is rebooting it
programmatically. It isn't crashing and rebooting.

dmknutsen, 2021-04-20:
I agree.  Worth note is that the "reboot()" actually came from my test script in the MM case (FSW stayed running long enough to get to the end of the script) - which is why we were able to get the additional information.  Basically, we are getting the stdout data in a not so great way and I believe it is getting lost in the other test cases (MD and the 'test_app' cases) where it may be crashing and rebooting.

jphickey, 2021-04-20:
Just noticed that this is on a very old baseline.  The locking error in the trace is very reminiscent of an OSAL bug that was discovered and fixed months ago (original issue reported in nasa/osal#645, fixed in IC nasa/osal#750, dev build 184).

Are there plans to update to a newer baseline?

jphickey, 2021-04-20:
I would really like to see if this can be duplicated using the current baseline and only the framework apps (by e.g. deleting SAMPLE_APP or something).  Then we can debug that.  If it can be reproduced via the bundle code then we can better debug it.  Otherwise if it doesn't occur in the framework bundle baseline then we have a good data point to start from and see where it does break.  But I'm not sure if its worth debugging on a 6 month old baseline that has hundreds of known issues already solved....

skliper, 2021-04-20:
Lets put this on hold until we get the updated baseline and see if it can be recreated with the latest eval branch.

skliper, 2021-09-06:
Latest case of this issue turned out to be a kernel/build environment mismatch.  Not a software issue, closing as invalid.

closed 2021-09-06 by skliper