nasa/cFE #1383: If an application fails to initialize; cFE will attempt to restart it indefinitely

dmknutsen, 2021-04-19:
**Is your feature request related to a problem? Please describe.**
Outside of issuing a processor reset - I'm not sure there is a clean way to exit this scenario (which in theory should never happen in flight).  This scenario would be further complicated if the app attempting to be started/restarted is included in the startup script.

**Describe the solution you'd like**
Consider implementing a maximum number of restarts.

**Requester Info**
Dan Knutsen
NASA Goddard

skliper, 2021-04-19:
This is for the case where the app is set to restart on exception, and an exception occurs as part of startup.  At least I think this is the scenario (please correct if I'm mistaken).

dmknutsen, 2021-04-20:
Yep - that is correct