nasa/cFE #1394: Define file subtype value behavior/expectations

skliper, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
Not clear if the CFE_FS_InitHeader SubType needs to be one of the FS enums or if it can be user defined by apps.  Note there is no longer a shell file created by ES:
https://github.com/nasa/cFE/blob/e80aae94e0f56b868657daba965c590766a4dc57/modules/core_api/fsw/inc/cfe_fs_extern_typedefs.h#L101-L108

**Describe the solution you'd like**
Need to determine if FS should define all file subtypes, or treat it as an extendable field (or whatever).  That will affect if the SHELL subtype gets removed or renamed (since there is still an app that would create it).  Note right now apps don't even use `CFE_FS_InitHeader`, but they do currently set unique values.

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-04-21:
Could add context (app name) to the header to deconflict subtype when extended.

astrogeco, 2021-04-21:
[CCB:2021-04-21](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-April-21-2021) DISCUSSED

skliper, 2021-04-21:
The SubTypes defined within the cFE FS header are cFS framework only, and apps are allowed to extend.  Could be future work issue to add app context to the header to avoid conflict.

PR #1575 (2021-05-27): Fix #1394, Remove shell file subtype and clarify scope
**Describe the contribution**
Fix #1394 - removes `CFE_FS_SubType_ES_SHELL` and clarifies the scope of `CFE_FS_SubType_Enum_t` to just include the cFE defined file types

**Testing performed**
Build and run unit tests, passed

**Expected behavior changes**
None, wasn't used since shell command was removed from CFE/ES.

**System(s) tested on**
 - Hardware: Intel I5/Docker
 - OS: Ubuntu 18.04
 - Versions: Bundle main + this commit

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consi [... cut]
commit 5febad7d (2021-05-27): Merge pull request #1575 from skliper/fix1394-file_subtype

Fix #1394, Remove shell file subtype and clarify scope
commit 6ce02300 (2021-05-27): Fix #1394, Remove shell file subtype and clarify scope
PR #1584 (2021-06-08): cFE Integration candidate: 2021-06-01
## Context

Part of <https://github.com/nasa/cFS/pull/263>

## Tests

cFE Checks <https://github.com/nasa/cFE/pull/1584/checks>
cFS Bundle Checks <https://github.com/nasa/cFS/pull/263/checks>

Note CodeQL failure in cFE checks because workflow points at "main". This is resolved by the cFS-bundle IC.

## Authors

@pepepr08 
@ArielSAdamsNASA 
@jphickey 
@skliper 

## Description

### PR #1580 

Fix #1364, Update ES verify errors to match 

Ensures error messages in `cfe_es_verify.h` match test logic. 

### PR #1578 

Fix #1366, Resolve mismatched endforeach  [... cut]
closed 2021-06-08 by astrogeco