nasa/cFE #1395: Scrub parameter documentation section of command codes

skliper, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
Parameter documentation for command codes not complete or always up to date

**Describe the solution you'd like**
Really the structure is the preferred place to document the parameters, the command code documentation is probably better off just referencing the structure.  Need to confirm all the information in the command code documentation is in the structure documentation and then clean.

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-07-28:
As part of scrubbing command docs in #1396 I also reviewed the documentation on the structures themselves.

In short - I don't see any major issues here.

In some cases structures are re-used between commands, and the structure documentation is therefore more generic in nature, where the command documentation gets into specifics about how its used.  For instance EVS has a generic command that accepts a bitmask, and the `CFE_EVS_BitMaskCmd_Payload_t` just defines it as follows:

https://github.com/nasa/cFE/blob/33a4f19ce6dc2c132823b3c8c487a59d5b4b81a8/modules/evs/fsw/inc/cfe_evs_msg.h#L1020

However in the command documentation it says how the specific bits are defined for that particular command.

https://github.com/nasa/cFE/blob/33a4f19ce6dc2c132823b3c8c487a59d5b4b81a8/modules/evs/fsw/inc/cfe_evs_msg.h#L540-544

So in my opinion this is appropriate - I don't see anything I feel compelled to change here.

jphickey, 2021-07-28:
The PR for related issue #1396 also updates the description/verification sections where they were lacking, so it can be associated with this issue too (discussed in CCB 2021-07-28).

PR #1719 (2021-07-28): Fix #1396, scrub command documentation
**Describe the contribution**
Update documentation for all command codes in CFE core apps, in particular the error conditions, to confirm it matches the
actual implementation.

For common operations that apply to all commmands (length verification) this is now put into the users guide as a general statement rather than repeating the info in every command.

Fixes #1396 
Fixes #1395 

**Testing performed**
Build documentation and confirm output

**Expected behavior changes**
None (docs only)

**System(s) tested on**
Ubuntu

**Contributor Info - All informat [... cut]
closed 2021-09-24 by skliper