nasa/cFE #1398: `CFE_EVS_EventType_Enum_t` only used in 1 location, no `CFE_EVS_EventID_t` type...

skliper, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
Numerous locations where EventType is declared as uint16, but there's a defined type `CFE_EVS_EventType_Enum_t`

**Describe the solution you'd like**
Use CFE_EVS_EventType_Enum_t everywhere

**Describe alternatives you've considered**
Use uint16 everywhere

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-04-21:
This allows the label to be printed for the value (naming convention), so it makes sense to have this for ENUM but not for general types.

astrogeco, 2021-04-21:
[CCB:2021-04-21](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-April-21-2021) Discussed

PR #2308 (2024-12-10): Fix #1398, Consistent use of `CFE_EVS_EventType_Enum_t` for `EventType`
**Checklist**
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fixes #1398
  - Remaining `EventType` parameters/variables declared explicitly as `int16` converted over to use the defined type

Note: https [... cut]
closed 2024-12-10 by dzbaker