nasa/cFE #1417: Generate new set of CFE coverage test stubs

jphickey, 2021-04-22:
**Is your feature request related to a problem? Please describe.**
To achieve max flexibility the CFE coverage test stubs should have no built-in logic, with all handler logic in a separate function.

This is the pattern that was introduced with nasa/osal#966, along with the stub generator script.

**Describe the solution you'd like**
Run the stub generator script on CFE core API headers to generate a new set of "pure" stub implementation files with no extra logic in them.
Migrate any existing post-hook logic in the old stub to be a default handler function, as was done for OSAL stubs.

**Additional context**
Extension of original request in nasa/osal#832, but for CFE.  Depends on nasa/osal#966 to be accepted/merged before this can be done.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-04-22:
Ping @asgibson for awareness, this will be the next step in getting all the stubs updated to work in the preferred manner.

skliper, 2021-04-30:
Does this also fix #1393 and #1387?  If so, link.

jphickey, 2021-04-30:
Yes this fixes #1387 but does not fix #1393.  I can do an add-on PR for that part though.

asgibson, 2021-04-30:
I have pulled this update and run with cFS https://github.com/nasa/cFE/commit/c8b5e00bb722d8f59787ae8f82634feba5d39f60 with the only change being the use of cFE https://github.com/nasa/cFE/pull/1463.  All the CF unit tests run as they did before with no discernible difference.  As this is the goal, that is success.

One note: None of the new "handler" override of any cFE stub was attempted.  This only shows that whatever cFE API methods that are called by CF unit tests still operate as expected.  This is not extensive, but several are called during testing.

Some of those used (not a complete list):

- CFE_EVS_SendEvent
- CFE_TBL_ReleaseAddress
- CFE_TBL_GetAddress
- CFE_MSG_GetFcnCode
- CFE_MSG_GetSize
- CFE_MSG_SetMsgTime
- CFE_TIME_GetTime
- CFE_SB_TransmitMsg

commit 972516c6 (2021-05-11): Fix #1417, generated coverage stubs for CFE core

Update CFE core stub libraries to use generated stubs, using the generator
script now part of UT Assert.  All existing stub logic is converted to
a default handler routine.

This also entailed moving all of the internal API prototypes to be
in the "core_private" interface lib rather than in "core_api".  The
only reason this was not done earlier was due to issues with the stubs,
but that is alleviated when using generated stubs.
commit fc766c5e (2021-05-11): Merge pull request #1463 from jphickey/fix-1417-cfe-coverage-stubs

Fix #1417, #1387, #1393, generated coverage stubs for CFE core
PR #1492 (2021-05-12): cFE Integration candidate: 2021-05-11
## Context 

Part of <https://github.com/nasa/cfs/pull/256>

## Testing 

cFE Checks <https://github.com/nasa/cFE/pull/1492/checks>
cFS Bundle Checks <https://github.com/nasa/cFS/pull/256/checks>

## Description

### PR #1487

 Fix #1486, Remove broken travis-ci script 

### PR #1463

Fix #1417, generated coverage stubs for CFE core

Updates CFE core stub libraries to use stubs generated from new UT Assert script. Converts existing stub logic to
a default handler routine.

Moves all of the internal API prototypes from "core_api" to the "core_private" inter [... cut]
closed 2021-05-12 by astrogeco