nasa/cFE #1432: Include element that caused write error in event reporting in `CFE_EVS_WriteAppDataFileCmd`

skliper, 2021-04-28:
**Is your feature request related to a problem? Please describe.**
Element that caused error not included in event, so not obvious how much of the write was completed
https://github.com/nasa/cFE/blob/73c338d544f85cb0c1a97b743d3ae9f7a35039b5/modules/evs/fsw/src/cfe_evs_task.c#L1746-L1748

**Describe the solution you'd like**
Add the element number

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2023-05-02:
@skliper Were you thinking something like this?
[image]
or
[image]

thnkslprpt, 2023-05-03:
@skliper - how would you get the second value (of)?

Isn't the total number of entries that was supposed to be written only known after the `for` loop ends? Or do you mean to just use `CFE_PLATFORM_ES_MAX_APPLICATIONS`?

commit 1dfc7492 (2024-08-02): Merge pull request #2326 from thnkslprpt:fix-1432-add-element-to-write-error-event

Fix #1432, Add element to write error event in CFE_EVS_WriteAppDataFileCmd()
commit 5b53ca27 (2024-08-02): Fix #1432, Add element to write error event in `CFE_EVS_WriteAppDataFileCmd`
PR #2326 (2024-08-02): Fix #1432, Add element to write error event in `CFE_EVS_WriteAppDataFileCmd()`
**Checklist**
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fixes #1432 
  - Adds element number to the event that reports a write error in `CFE_EVS_WriteAppDataFileCmd()` to provide more contex [... cut]
closed 2024-08-02 by dzbaker