nasa/cFE #1434: Replace magic number in `CFE_EVS_MAX_PORT_MSG_LENGTH` defnition

skliper, 2021-04-28:
**Is your feature request related to a problem? Please describe.**
Magic number used:
https://github.com/nasa/cFE/blob/73c338d544f85cb0c1a97b743d3ae9f7a35039b5/modules/evs/fsw/src/cfe_evs_task.h#L63

**Describe the solution you'd like**
Use a defined size (this is probably for time)

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2024-01-16:
@skliper Can the magic number (30) just be replaced with `CFE_TIME_PRINTED_STRING_SIZE`?

skliper, 2024-01-16:
My guess is it's probably a little bigger than that due to the other information in the printed event.  Example:
```
EVS Port1 1980-012-14:03:20.65013 66/1/SC 9: SC Initialized. Version 3.1.99.0
```
It doesn't look like these parts would be accounted for:
`EVS port1 `
`66/1`
`9: `

thnkslprpt, 2024-07-14:
Note: see https://github.com/nasa/cFE/pull/2566 and https://github.com/nasa/cFE/issues/2571