nasa/cFE #1446: Remove defines for EVS LogMode

skliper, 2021-04-29:
**Is your feature request related to a problem? Please describe.**
Enum is defined here:
https://github.com/nasa/cFE/blob/c8b5e00bb722d8f59787ae8f82634feba5d39f60/modules/core_api/fsw/inc/cfe_evs_extern_typedefs.h#L66-L81

Defines here:
https://github.com/nasa/cFE/blob/c8b5e00bb722d8f59787ae8f82634feba5d39f60/modules/evs/fsw/inc/cfe_evs_msg.h#L910-L912

**Describe the solution you'd like**
Remove defines, replace any references with the enums.

**Describe alternatives you've considered**
Boolean for overwrite (although cmd/tlm should be fixed size)

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-05-17:
Ping @pepepr08

pepepr08, 2021-05-18:
I'll work on this

PR #1553 (2021-05-27): Fix #1446, Remove unused EVS LogMode defines
**Describe the contribution**
Removed unused EVS LogMode defines and references to these defines.

Fixes #1446 

**Testing performed**
Steps taken to test the contribution:
1. Build and run unit tests

**Expected behavior changes**
There are no behavior changes. Everything continues to run the same.

**System(s) tested on**
 - OS: Ubuntu 18.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Jose F. Martinez Pedraza, NASA GSFC code 582
commit 0cab138f (2021-05-27): Merge pull request #1553 from pepepr08/fix1446-logmode-defines

Fix #1446, Remove unused EVS LogMode defines
commit 6de6d242 (2021-05-27): Fix #1446, Remove unused EVS LogMode defines
PR #1584 (2021-06-08): cFE Integration candidate: 2021-06-01
## Context

Part of <https://github.com/nasa/cFS/pull/263>

## Tests

cFE Checks <https://github.com/nasa/cFE/pull/1584/checks>
cFS Bundle Checks <https://github.com/nasa/cFS/pull/263/checks>

Note CodeQL failure in cFE checks because workflow points at "main". This is resolved by the cFS-bundle IC.

## Authors

@pepepr08 
@ArielSAdamsNASA 
@jphickey 
@skliper 

## Description

### PR #1580 

Fix #1364, Update ES verify errors to match 

Ensures error messages in `cfe_es_verify.h` match test logic. 

### PR #1578 

Fix #1366, Resolve mismatched endforeach  [... cut]
closed 2021-06-08 by astrogeco