nasa/cFE #1448: Simplify LogOverflowCounter increment logic

skliper, 2021-04-29:
**Is your feature request related to a problem? Please describe.**
The CFE_EVS_Global.EVS_LogPtr->LogOverflowCounter gets incremented in two locations and is done based on the LogFullFlag:
https://github.com/nasa/cFE/blob/c8b5e00bb722d8f59787ae8f82634feba5d39f60/modules/evs/fsw/src/cfe_evs_log.c#L52-L64

**Describe the solution you'd like**
Pull up higher and change the if/else statement to:
`if(CFE_EVS_Global.EVS_LogPtr->LogMode != CFE_EVS_LogMode_DISCARD)`

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2023-04-25:
@skliper - Unless I'm misreading the logic here, most of the code in the `else` block, from the `memcpy` down, is also executed in the following path/circumstances:
`LogFullFlag == false`
`LogMode == CFE_EVS_LogMode_DISCARD`

Is this intentional?

Executing that block with the `memcpy` etc. based on `if(CFE_EVS_Global.EVS_LogPtr->LogMode != CFE_EVS_LogMode_DISCARD)` would therefore not be logically equivalent - it would miss that path/case mentioned above.

skliper, 2023-04-25:
No behavior change intended.  I think the code review comment was suggesting something like the following pseudocode:
```
if (LogFull)
  Increment log full flag

if (!LogFull || DISCARD)
  The rest of the logic to add the event to the log and count it
```

thnkslprpt, 2023-04-26:
Yes but again that is the issue I see - I'm wondering whether it's intentional or not.
From what I can tell, the logically equivalent pseudocode (to the current code) would be:
```
if (LogFull)
  Increment log full flag

if (!LogFull || !DISCARD)
  The rest of the logic to add the event to the log and count it
```

_Note the `!` before `DISCARD`._
The above code changes, when implemented, passes all tests.

That's why I'm wondering whether the rest of the code (to add the event data to the log etc.) is actually supposed to occur in the case of `LogFull && !DISCARD` - which is what happens now.

Hopefully this makes sense.

skliper, 2023-04-26:
@thnkslprpt - Good catch!  Definitely  `!DISCARD`.  Want to add to the log either if the log isn't full or if not set to discard mode.  The `Logfull && !DISCARD` would only add to the log if it's full (which wouldn't ever happen if you can't add to the log when it's not full).

thnkslprpt, 2023-04-26:
Cool no worries - hopefully the PR makes it a little clearer for future maintenance :sweat_smile:

PR #2309 (2024-12-10): Fix #1448, Simplify and clarify `EVS_AddLog` logic
**Checklist**
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
- Fixes #1448
  - Removes duplication of incrementing the `LogOverflowCounter`
  - Simplifies the function and makes logical flow a little clearer

**Testing perfor [... cut]
closed 2024-12-10 by dzbaker