nasa/cFE #1460: Combine mostly duplicated logic in `CFE_SB_BroadcastBufferToRoute`

skliper, 2021-04-29:
**Is your feature request related to a problem? Please describe.**
Duplicated logic:
https://github.com/nasa/cFE/blob/c8b5e00bb722d8f59787ae8f82634feba5d39f60/modules/sb/fsw/src/cfe_sb_api.c#L1650-L1667

**Describe the solution you'd like**
Factor out, make an else with shared logic and internal if for the unique logic

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2022-09-24:
You're thinking something like this Jake (@skliper)?
https://github.com/nasa/cFE/blob/3d8829ecb98343552c2bb66cf1d45e9b83420930/modules/sb/fsw/src/cfe_sb_api.c#L1661-L1678

thnkslprpt, 2022-09-26:
Cheers Jake.

commit 3d8829ec (2022-10-03): Fix #1460, Duplicated Logic
commit c363f19d (2022-10-03): Merge pull request #2151 from thnkslprpt/fix-1460-duplicated-logic

Fix #1460, Duplicated Logic in CFE_SB_BroadcastBufferToRoute
PR #2161 (2022-10-03): cFE Integration Candidate: Caelum-rc4+dev23
**Describe the contribution**
PR #2145
- Fixes issue #2144- Propagate CMAKE_EXPORT_COMPILE_COMMANDS variable

PR #2148
- Fix #2147, Remove CodeQL Paths Ignore #2148

PR #2151
- Fix #1460, Duplicated Logic in CFE_SB_BroadcastBufferToRoute

PR #2156
- Fix #1540, Remove 'return;' from last line of void functions.

PR #2154
- Fix #2153, Remove unnecessary parentheses around return values.

**Testing performed**

cFE checks <https://github.com/nasa/cFE/pull/TBD/checks>
cFS Bundle checks <https://github.com/nasa/cFS/pull/TBD/checks>

**Expected behavior ch [... cut]
PR #2151 (2022-10-03): Fix #1460, Duplicated Logic in CFE_SB_BroadcastBufferToRoute
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFE/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #1460
Removes duplicated logic in the final if/else in CFE_SB_BroadcastBufferToRoute.

**Testing performed**
Standa [... cut]
closed 2022-10-03 by dzbaker