nasa/cFE #1462: Decouple SB and SBR in coverage tests

jphickey, 2021-04-30:
**Is your feature request related to a problem? Please describe.**
Currently the SB coverage test is not like the others in that it links with the real SBR sub-module, rather than a stub version.  This becomes evident when generating the stubs as in #1417 

**Describe the solution you'd like**
- Make SBR unit tests follow the pattern of all other modules - Generate a set of stubs for SBR using the generator script and it should use the new cmake functions to add the coverage test.
- Update the SB unit tests where it was depending on having a real SBR implementation (should be just a matter of setting the intended stub output where it was missing)

**Additional context**
Per the system architecture/design, Each sub-module should be individually coverage tested, using stubs for all other modules.  This is an exception to that general design because SBR does not have proper stub functions for testing dependencies in isolation (e.g. SB in this case) like all other modules do.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2021-04-30:
Note original issue was #962, closed in favor of this one.

jphickey, 2021-07-16:
Note that the fact that SB and SBR are coupled together during coverage test means certain features of SB might not be reachable in the coverage test.  As noted in #1668, collisions are not possible when linked with the "direct" SBR implementation, so collision handling in SB cannot be covered:

https://github.com/nasa/cFE/blob/2afdbc16e360530a9b4bbe4e0dcc814f675c1acb/modules/sb/fsw/src/cfe_sb_api.c#L1125-L1130

Splitting SBR from SB for coverage test, and providing stubs for SBR functions (like other modules) would likely improve coverage test capability.