nasa/cFE #1467: Report CFE_ES_WaitForSystemState timeouts

skliper, 2021-04-30:
**Is your feature request related to a problem? Please describe.**
Timeouts aren't reported when using `CFE_ES_WaitForSystemState`, so not obvious when/if synchronization isn't properly occurring.
Example (although it's the same in all services):
https://github.com/nasa/cFE/blob/c8b5e00bb722d8f59787ae8f82634feba5d39f60/modules/sb/fsw/src/cfe_sb_task.c#L87

**Describe the solution you'd like**
Report timeout and possible system health issues.

**Describe alternatives you've considered**
Could be considered an error, may end up with strange behavior.

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2023-04-06:
@skliper Do you think it's useful to include the MinSystemState and/or TimeOutMilliseconds parameters in the syslog message or just something simple like:

```
        else
        {
            Status = CFE_ES_OPERATION_TIMED_OUT;
            CFE_ES_WriteToSysLog("%s(): Timeout reached.", __func__,);
            break;
        }
```

skliper, 2023-04-06:
I like verbosity in the log, MinSystemState, current system state, and TimeOutMilliseconds sounds good!