nasa/cFE #1483: Add a CI action to build and execute the cFE coverage tests

skliper, 2021-05-04:
**Is your feature request related to a problem? Please describe.**
CI doesn't build and execute the cFE coverage tests

**Describe the solution you'd like**
Add

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

skliper, 2021-05-21:
@pepepr08 - want to take this one?  Trigger on push or pull_request and include the duplicate check.  Should be able to just `make -C pathtomodule` and then `make -C pathtomodule test` for each module.

pepepr08, 2021-05-24:
Yes   I can work on this.

astrogeco, 2021-05-24:
@skliper aren't we building these at the bundle level?

skliper, 2021-05-25:
> @skliper aren't we building these at the bundle level?

@astrogeco Yes, but that means they don't get tested for cFE PR's at the time of submittal, so errors sneak in and slow the merge process.  Also will make coverage verification easier at the level of cFE since it'll report just the cFE coverage vs the entire bundle.

PR #1590 (2021-06-10): Fix #1483, Add CI workflow to run cFE coverage tests
**Describe the contribution**
Fix #1483
Run cFE coverage tests for every push and pull request. Use a checks-for-duplicates job to avoid duplicate runs of the CI Action.

**Testing performed**
Steps taken to test the contribution:
1. Pushed to fork, CI action ran successfully: https://github.com/pepepr08/cFE/actions/runs/886331016
1. Created a temporary branch and introduced an ES bug to force a workflow failure. Workflow failed successfully: https://github.com/pepepr08/cFE/actions/runs/886371053
1. Reverted commit that introduced bug to ver [... cut]
commit 06a218c5 (2021-06-10): Fix #1483, Install lcov in workflow
commit 569a646b (2021-06-10): Merge pull request #1590 from pepepr08/fix1483-ci-lcov

Fix #1483, Add CI workflow to run cFE coverage tests
commit 99b1336f (2021-06-10): Fix #1483, make sure job doesn't run if duplicate
closed 2021-06-16 by astrogeco