nasa/cFE #1485: CFE_ES_PoolCreateEx NumBlockSizes error handling

zanzaben, 2021-05-06:
**Describe the bug**
In the header file of CFE_ES_PoolCreateEx, for parameter NumBlockSizes it says "If set equal to zero or if greater than 17, then default block sizes are used." In the code though if NumBLockSizes is greater than CFE_PLATFORM_ES_POOL_MAX_BUCKETS (which is set to 17) then it returns error code CFE_ES_BAD_ARGUMENT.

**Expected behavior**
The header and functionality should match. 

**Reporter Info**
Alex Campbell GSFC

jphickey, 2021-05-06:
I would vote to make the header/documentation match what the implementation does.  Passing in a bad value should be an error.

zanzaben, 2021-05-06:
That's fine but the implementation for when NumBLocksSizes equals 0 is weird. It will be set to the default only when BlockSizes is null, and will stay as 0 when BlockSizes isn't null.

jphickey, 2021-05-06:
> That's fine but the implementation for when NumBLocksSizes equals 0 is weird. It will be set to the default only when BlockSizes is null, and will stay as 0 when BlockSizes isn't null.

Yeah, it is kinda weird, but I vaguely recall preserving that "feature" of the API (only use default set if the pointer is NULL, regardless of NumBlockSizes).  Didn't want to break any code that might be depending on that.

I wouldn't want to change the behavior at this point - we should keep it as being the BlockSizes pointer (NULL or not NULL) that controls whether defaults are used.

As for the case of a `BlockSizes != NULL`  and `NumBlockSizes == 0`, does this have any major issues?  Obviously its a bad config and I don't expect it to do anything useful in that case, but so long as it doesn't crash or otherwise break the system I don't see it as being a major issue.  We could consider returning BAD ARGUMENT in that case too, but I don't see it as critical (maybe others have a different opinion).

skliper, 2021-05-06:
Let's just update the documentation to match the behavior for now (Caelum), feel free to open a future work issue to make the implementation make more sense.

PR #1562 (2021-05-27): Fix #1485, Document CFE_ES_PoolCreateEx NumBlockSizes error handling
**Describe the contribution**
Fix #1485 - updated documentation to match implementation

**Testing performed**
CI - documentation only

**Expected behavior changes**
None

**System(s) tested on**
CI

**Additional context**
None

**Third party code**
None

**Contributor Info - All information REQUIRED for consideration of pull request**
Jacob Hageman - NASA/GSFC
commit ee19f189 (2021-05-27): Fix #1485, Document CFE_ES_PoolCreateEx NumBlockSizes error handling
commit efd72a04 (2021-05-27): Merge pull request #1562 from skliper/fix1485-numblocksize_error_doc

Fix #1485, Document CFE_ES_PoolCreateEx NumBlockSizes error handling
closed 2021-06-08 by astrogeco