nasa/cFE #1504: Refactor TBL to reduce complexity, improve maintainability, reduce technical debt

skliper, 2021-05-11:
**Is your feature request related to a problem? Please describe.**
TBL functions are complex and utilize many modes.  Complex enough that minor maintenance  is avoided due to the high likelihood of breaking something.

**Describe the solution you'd like**
Break down large complex functions.

**Describe alternatives you've considered**
None

**Additional context**
Code review, related to #1493 in that sharing isn't clearly defined

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2022-11-02:
I have a couple questions, if someone has a minute to answer them.

Point 1: Why is `status` set to `CFE_TBL_ERR_REGISTRY_FULL`? It seems all that was checked for is whether the table was found or not, not if the registry was full.

Point 5: Is the preferred method to factor out multiple returns
a) variable `status` and check if `status == CFE_SUCCESS` before executing each block of code (i.e. if `status` has previously been set to an error state, it will skip all following blocks of code to the end)
b) put everything in a `while (true)` loop and `break` out where there are currently early returns
c) something else?

dzbaker, 2022-11-02:
> I have a couple questions, if someone has a minute to answer them.
>
> Point 1: Why is `status` set to `CFE_TBL_ERR_REGISTRY_FULL`? It seems all that was checked for is whether the table was found or [...]

@jphickey Would you know the answers?

jphickey, 2022-11-02:
This is a big undertaking here -- The problem is that TBL needs a significant overhaul to reduce the complexity of functions.  As a guideline the cyclomatic complexity of functions (based on the number of decision points and alternate paths that can be followed) should be kept less than 10.   Keeping the functions simple make them automatically more testable, easier to code review,  less likely to break things due to unexpected consequences, etc.  The biggest offender here by far is `CFE_TBL_Register` which has a fairly absurd amount of branches, followed by `CFE_TBL_Load`, but there are many others as well.

The point is not to just make the returns more readable by replacing with a different style, but to split up what is currently a big monolithic function into smaller parts.  

The preference is that there are no multiple returns at all after a refactoring - with small/simple functions this becomes easier to achieve.  (Except for a possible return early at the beginning as a result of argument/input checks, but nothing in the middle during the business logic of a function - functions should either fail immediately, before any action happens, or run to the end).

That being said, the `CFE_TBL_ERR_REGISTRY_FULL` status code seems correct, that code is first checking if the table is already registered and if not, it calls `CFE_TBL_FindFreeRegistryEntry()`, and if there _still_ no defined index after that, it must mean the registry was full.  

But this is a great example of hard-to-follow logic buried in the middle of a long function - it is not obvious what's going on here, and this function is over 300 (!!) lines long.  The code is very suspicious-looking, because it will enter the next "if" statement (because `Status != CFE_TBL_WARN_DUPLICATE` evaluates as true), causing it to find a free handle, but throw it away because it eventually skips the rest of the logic.

In short, it's really confusing, and really hard to follow, and nobody wants to touch this code because the chances of unintended consequences are extremely high - for example if that `CFE_TBL_ERR_REGISTRY_FULL` status code were simply classified as a warning instead of an error, this code would segfault and crash - a completely non-obvious side effect of an innocent-seeming change.

It is this type of possible non-obvious side effects that needs to be resolved by refactoring the code to make it simpler and cleaner.