nasa/cFE #1533: Convert leap seconds command parameter to unsigned

skliper, 2021-05-17:
**Is your feature request related to a problem? Please describe.**
`int16` type used for `LeapSeconds` in command payload:
https://github.com/nasa/cFE/blob/84ba9a9974794e239b989cdc4e2359216e44fab0/modules/time/fsw/inc/cfe_time_msg.h#L749-L752

**Describe the solution you'd like**
Convert to `uint16`.

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

zanzaben, 2021-05-17:
Leap Seconds can technically be negative. So far everyone in history has been to add a second so it is incredibly unlikely that they will overall be negative but not impossible.

astrogeco, 2021-05-17:
I vote we keep the signed int: 

> [...] if the Earth's rotation keeps accelerating, as it has done in 2020, the IERS may decide to announce the world's first negative leap second at some point in the future.

https://www.timeanddate.com/time/leapseconds.html

skliper, 2021-05-17:
Concur, closed as invalid.

closed 2021-05-17 by skliper