nasa/cFE #1534: Replace if's with table and loop in `CFE_TIME_GetClockInfo`

skliper, 2021-05-17:
**Is your feature request related to a problem? Please describe.**
`CFE_TIME_GetClockInfo` could just be a table/loop instead of individual `if`'s

**Describe the solution you'd like**
Simplify with table/loop

**Describe alternatives you've considered**
None

**Additional context**
Code review

**Requester Info**
Jacob Hageman - NASA/GSFC

thnkslprpt, 2022-10-25:
@skliper Jake can you provide an example of the type of structure you had in mind - just want to make sure I understand you correctly here :-)
Cheers

skliper, 2022-10-25:
Another paraphrase from code reviews.

I think the concept here was to create a table of checks and the bit to set, and just loop through it.  Basically create a table with the value to test, the value to test against, and the bit to set if that test is true.  Then just loop through the table to do the tests and set the bits.

Something like:

```
typedef struct
{
  uint32 StateToTest;
  uint32 ValueToTest;
  uint32 SetIfTrue;
} StateTestStruct_t;

StateTestStruct_t StateTestStruct[] = 
  { 
    {RefState->ClockSetState, CFE_TIME_SetState_WAS_SET, CFE_TIME_FLAG_CLKSET}, 
    {...entry for each state
  }

then just for loop through the array and if StateToTest == ValueToTest then StateFlags |= SetIfTrue.
```

Up to @dmknutsen and @dzbaker on the priority of this one... it's just a restructure (same functionality).

thnkslprpt, 2022-10-25:
Thanks for the clarification