nasa/cFE #1571: Convert or remove "cfe requirements .docx"

astrogeco, 2021-05-24:
**Is your feature request related to a problem? Please describe.**

`cfe requirements.docx` is not easily readable since it requires both a download and the user to have MS Word or equivalent installed. 

The document's purpose is ambiguous given the existence of `cFE_FunctionalRequirements.csv`. I've also seen some "TBDs" in the document hinting that there is some more work needed. 

**Describe the solution you'd like**

Various options

Delete `cfe requirements.docx` and add a readme to the docs directory that explains the different files. Additionally,

1. Convert to Markdown *or*
2. Convert to a pdf 

**Describe alternatives you've considered**
None

**Additional context**
None

skliper, 2021-05-25:
`cfe requirements.docx` documents the cFS system level requirements, vs the functional level requirements that are documented in `cFE_FunctionalRequirements.csv`.

astrogeco, 2021-05-25:
What do you think of converting it to Markdown? For the young and naive among us, what would be a good way of explaining the difference between functional and system requirements?

skliper, 2021-05-25:
Convert to markdown would be a definite improvement.  System requirements are higher level, often conceptual description of what the system is required to do.  Functional requirements are derived from the system requirements and define explicit, testable behavior.  We have a functional test to verify every functional requirement.