nasa/cFE #1617: UT_DefaultHandler_CFE_SB_ReceiveBuffer does not adequately model behavior of CFE_SB_ReceiveBuffer for unit test purposes

asgibson, 2021-06-11:
**Describe the bug**
UT_DefaultHandler_CFE_SB_ReceiveBuffer only sets the value at BufPtr when Status is => 0 (CFE_SUCCESS == 0); however CFE_SB_ReceiveBuffer changes the value at BufPtr to NULL whenever the Status value is _not_ CFE_SUCCESS.It is possible to get a NULL with a non-zero positive result when set by the user.  It is not possible to get the NULL value from the handler with a negative Status, a behavior that the production code has.  There is no guarantee that the CFE_SB_Buffer_t **BufPtr passed into the function does not already contain a non-NULL value from the code under test (CUT).

**To Reproduce**
Steps to reproduce the behavior:
1. Write a unit test for some CUT that uses the function and do not set a new handler
2. Make it so a negative value is received back from the function (i.e. CFE_SB_TIME_OUT)
3. If the CUT does not do it automatically, temporarily change the passed in address to have a value at that address
4. Test results show that the expected NULL value at the given argument is unchanged

**Expected behavior**
The follow-on logic from the default handler should match the logic in the production code, OR
Redesign the stubs to have no follow-on logic so unit test developer is in complete control of outside effects.

**Code snips**
https://github.com/nasa/cFE/blob/a16c78e1853c9b46ce5d1efae882ea5bf3a170cf/modules/core_api/ut-stubs/src/cfe_sb_handlers.c#L197-L209
https://github.com/nasa/cFE/blob/a16c78e1853c9b46ce5d1efae882ea5bf3a170cf/modules/sb/fsw/src/cfe_sb_api.c#L2015-L2019

**System observed on:**
 - RHEL-7.8

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Alan Gibson
NASA  GSFC/587

skliper, 2021-06-15:
Is this not already solved by supporting override of the default handlers?  The trend for the default handlers is to model less of the real implementation vs more, so unique behavior can be owned by the test that requires that behavior.  Closing based on that approach and the capabilities already supplied, happy to set up a splinter if it helps.

asgibson, 2021-06-15:
The part that makes the CFE_SB_ReceiveBuffer stub different than other stubs, at least from what I have seen, is that the other stubs follow on logic seeks to mimic the production code.  I understand that handlers will be required in other instances, but this one does not seem to make any sense because the stub logic does not follow the production logic.  Why is the stub logic different?

skliper, 2021-06-15:
The default handlers are written to support the framework tests, you are welcome to use them if they suite the needs of your tests but if not you need to write your own.  This sort of request is out of scope, in that there are insufficient framework resources to update handlers to support every app test and that's really not the intent of the default handlers anyways.  They are absolutely inconsistent, but there are no resources identified at this point to fix them all since they have been deemed "good enough" for what the framework needs.  If/when resources are identified to address inconsistencies in the handlers or increasing the scope on handler support we will.

closed 2021-06-15 by skliper