nasa/cFE #1618: Add compile time assert that platform config sizes are < mission config sizes

skliper, 2021-06-11:
**Is your feature request related to a problem? Please describe.**
A stakeholder bumped up CFE_PLATFORM_SB_MAX_PIPES to 128 but left CFE_MISSION_SB_MAX_PIPES as 64, which is a broken config since tlm for PipeDepthStats is sized using CFE_MISSION_SB_MAX_PIPES.

**Describe the solution you'd like**
Add a compile time assert that CFE_PLATFORM_* is < CFE_MISSION_*

**Describe alternatives you've considered**
Charge more to fix user errors.

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC, ping @excaliburtb

jphickey, 2021-06-16:
Are we sure that this is is a configuration we need to restrict?  Obviously the user won't be able to get status on the items beyond the "MISSION" config, but maybe a user doesn't need that, and wants to keep their TLM messages small.

IMO the asserts should be used for configs that are truly broken (i.e. CFE won't run or will segfault/crash/run out of memory/etc).  If its just atypical but "runs as configured", then why do we need to limit user choice?

For reference: https://xkcd.com/1172/

skliper, 2021-06-16:
Maybe assert from within a configurable context (within the platform config settings file)?  I'd think the warning/error is useful for catching the most likely misconfiguration case but assert could be removed to support a use case where it was intentional.