nasa/cFE #1620: CFE_ES_QUERY_ONE_CC not performing as expected

dmknutsen, 2021-06-15:
**Describe the bug**
Within the CFE_ES_APP_TLM_MID packet - the following items are not behaving as expected:
1. BSSSize, CodeSize, and DataSize always report 0 for a given App.  
2. StartAddress for a given App changes whenever the software reset.

**To Reproduce**
1. Start FSW
2. Send the CFE_ES_QUERY_ONE_CC  for a given App
3. Verify that the BSSSize, CodeSize, and DataSize always report 0 + StartAddress is not fixed (this may be by design...I am thinking that it should only change if the software is re-compiled).

**System observed on:**
Oracle VM VirtualBox
OS: ubuntu-20.10
Versions: cfe v6.8.0-rc1+dev218, osal v5.1.0-rc1+dev109, psp v1.5.0-rc142

**Reporter Info**
Dan Knutsen
NASA/Goddard

skliper, 2021-06-15:
Is the AddressesAreValid parameter in the packet set as TRUE?  If it's FALSE, I recommend skipping those checks since it means the OS_ModuleInfo is likely not supported on the platform.

dmknutsen, 2021-06-15:
I just checked and it is set to false...we will go ahead and remove the checks.

closed 2021-06-15 by skliper