nasa/cFE #1628: Add code coverage test for infinite loop in cfe_es_api.c, CFE_ES_ExitApp

pepepr08, 2021-06-16:
**Is your feature request related to a problem? Please describe.**
Missing coverage for CFE_ES_ExitApp function that contains an infinite loop.

https://github.com/nasa/cFE/blob/d03e24e675fcd253222e40a5c544920e03854c5f/modules/es/fsw/src/cfe_es_api.c#L444-L464

**Describe the solution you'd like**
Add coverage tests to include part of the function that is currently not being tested.

**Describe alternatives you've considered**
None.

**Requester Info**
Jose F. Martinez Pedraza/NASA GSFC

skliper, 2021-06-16:
Maybe make the while loop dependent on system state instead of while(1), then it could be tested normally.

avan989, 2023-06-12:
code coverage is not possible without entering an infinite loop.