nasa/cFE #1631: Run continuous-integration workflows with the bundle integration-candidate branch where needed

astrogeco, 2021-06-17:
**Is your feature request related to a problem? Please describe.**
Some CI workflows checkout the cFS bundle, it would be good to see how some things fail against main but pass with IC

**Describe the solution you'd like**
Add a matrix build so workflows that depend on the bundle can test both. Add a badge to the readme. 

**Describe alternatives you've considered**

**Additional context**
Add any other context about the feature request here.

**Requester Info**
Full name and company/organization if applicable

skliper, 2021-06-17:
As an alternative you can push a bundle to your fork with any combination of branches you want (IC or otherwise) to see the bundle CI run results.  Or merge what is working at the bundle level into main, and rerun the PR IC?  Or have a manual trigger to run a specific PR against the bundle IC?  Running every PR against bundle IC might be overkill.

jphickey, 2021-06-18:
This is where some of the complexity of the old babelfish/bamboo workflows came from.

The rule was to look for matching branch names.  So, for instance, when testing a given branch, when assembling a bundle for CI purposes, it would look for a matching branch name in all other repos, falling back to the stable branch if the branch name didn't exist.

So, for instance, if a change required a concurrent modification to CFE and OSAL (a common paradigm), as long as it was pushed with the same branch name to both repos (for example, `ic-2021-06-18` if using date-based IC naming), the CI script would pick this branch for both CFE and OSAL, and use `main` for everything else.  (We had called it "development" back then, but same idea - it was the stable-ish branch that had the latest stuff but known to compile and run).