nasa/cFE #1644: Finish functional tests for cFE ES Miscellaneous APIs

zanzaben, 2021-06-29:
**Is your feature request related to a problem? Please describe.**
Need open source functional tests for certifiability

**Describe the solution you'd like**
This function got missed by the existing miscellaneous api test
– CFE_ES_ProcessAsyncEvent - Notification that an asynchronous event was detected by the underlying
OS/PSP.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Requester Info**
Alex Campbell GSFC

skliper, 2021-08-05:
Simply add a call in the existing test, it's void/void and just a wakeup so within the context of the functional test not much else to do.  Distribution/system testing on the user side confirms behavior based on implementation.

nmullane, 2021-08-05:
I'll work on this.

PR #1766 (2021-08-09): Fix #1644, finish ES misc API functional test
**Describe the contribution**
Fixes #1644 

**Testing performed**
1. Build and run cfe functional tests

**Expected behavior changes**
- No impact to behavior

**System(s) tested on**
 - Hardware: PC
 - OS: Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Niall Mullane - GSFC 582 Intern
commit 18a9295f (2021-08-09): Fix #1644, finish ES misc API functional test
commit 4c915652 (2021-08-09): Merge pull request #1766 from nmullane/fix1644-finish-es-misc-api-functional-test

Fix #1644, finish ES misc API functional test