nasa/cFE #1647: Add functional tests for cFE EVS Registration APIs

zanzaben, 2021-06-29:
**Is your feature request related to a problem? Please describe.**
Need open source functional tests for certifiability

**Describe the solution you'd like**
Add functional tests for cFE EVS Registration APIs
– CFE_EVS_Register - Register an application for receiving event services.
– CFE_EVS_Unregister - Cleanup internal structures used by the event manager for the calling Application.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Requester Info**
Alex Campbell GSFC

skliper, 2021-08-02:
The test application already utilizes a nominal success case of CFE_EVS_Register and uses the event system to report status.  CFE_EVS_Register error cases can clear the registration and are probably more appropriate in a "special" test app that doesn't utilize events or delays the checking until the system registered correctly.  Could do error checking (that resets the registration) and then reregister successfully, but this would clear any custom config and assumes an initial config (unless a helper function was added to eliminate duplication, but would still clear any filter modifications and counters).  Implementation is fully exercised via coverage tests, so not critical this round.

astrogeco, 2021-09-21:
@skliper do we need to close this for Caelum?

skliper, 2021-09-21:
@astrogeco I removed the Caelum milestone for the ones not critical this round.  The comment above provides justification.

> Implementation is fully exercised via coverage tests, so not critical this round.