nasa/cFE #1688: Add functional tests for cFE Get Table Information APIs

zanzaben, 2021-07-20:
**Is your feature request related to a problem? Please describe.**
Need open source functional tests for certifiability

**Describe the solution you'd like**
Add functional tests for cFE Get Table Information APIs
    CFE_TBL_GetStatus - Obtain current status of pending actions for a table.
    CFE_TBL_GetInfo - Obtain characteristics/information of/about a specified table.
    CFE_TBL_NotifyByMessage - Instruct cFE Table Services to notify Application via message when table requires management.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Requester Info**
Alex Campbell GSFC

nmullane, 2021-07-29:
I'll start working on this

nmullane, 2021-07-30:
### Do I need to test "complex" behavior of these functions
Currently, I'm just testing error conditions and the most basic successful calls of these functions. Should I test more complicated behavior of these functions?

Are there things in the following lists that I should be testing?
#### `CFE_TBL_GetStatus`:
* Test that pending update, validation statuses are returned at the correct time
  * Is there a way to trigger these states?
####  `CFE_TBL_GetInfo`:
* test number of users increments after the table is shared
* test crc is valid
* test last file loaded is correct after loading a file
* test table loaded once updates after loading
* test file created time makes sense

#### `CFE_TBL_NotifyByMessage`:
* Test messages are actually sent with correct message id, cc, parameter

skliper, 2021-07-30:
This first round through the priority is error conditions and whatever is easy to functionally check (it's more important to get the basic testing in for all the APIs, than a really high fidelity test for just a subset).  It is helpful to note in the test what you currently aren't covering and should be added eventually.

skliper, 2021-07-30:
Actually, instead of noting in the test the preference would be to add an issue that details the suggested additions.  Easier to track that way!

commit 7e854ff4 (2021-08-09): Fix #1688, Add table information functional tests
PR #1734 (2021-08-09): Add table api functional tests
**Describe the contribution**
A clear and concise description of what the contribution is.
- Fixes
  - Fix #1685
  - Fix #1688 
  - Fix #1686 
  - Fix #1687 
- Adds functional unit tests for
  - CFE_TBL_Register
  - CFE_TBL_Share
  - CFE_TBL_Unregister
  - CFE_TBL_GetStatus
  - CFE_TBL_GetInfo
  - CFE_TBL_NotifyByMessage
  - CFE_TBL_Load
  - CFE_TBL_Update
  - CFE_TBL_Validate
  - CFE_TBL_Manage
  - CFE_TBL_DumpToBuffer
  - CFE_TBL_Modified
  - CFE_TBL_GetAddress
  - CFE_TBL_GetAddresses
  - CFE_TBL_ReleaseAddress
  - CFE_TBL_ReleaseAddresses

**Testing performed [... cut]
PR #1772 (2021-08-11): cFE Integration candidate: 2021-08-10
**Describe the contribution**

PR #1737
- Fix #1733, Move global count into test global struct. 

PR #1722 
- Fix #806, Add ES application control API functional tests 
- Fix #1324, Add null pointer check to CFE_ES_ReloadApp functional test 

PR #1753 
- Fix #813, Add Generic Counter API test 

PR #1744 

- Fix #1743, update coverage test to use UtAssert macros 

PR #1734
  - Fix #1685
  - Fix #1688 
  - Fix #1686 
  - Fix #1687 

PR #1716 

- Fix #1645, Add resource ID API functional tests 

PR #1764 

- Fix #1763, last char truncated in coverage log outp [... cut]