nasa/cFE #1692: Add functional tests for cFE Miscellaneous Time APIs

zanzaben, 2021-07-20:
**Is your feature request related to a problem? Please describe.**
Need open source functional tests for certifiability

**Describe the solution you'd like**
Add functional tests for cFE Miscellaneous Time APIs
    CFE_TIME_Print - Print a time value as a string.
    CFE_TIME_Local1HzISR - This function is called via a timer callback set up at initialization of the TIME service.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Requester Info**
Alex Campbell GSFC

skliper, 2021-08-05:
CFE_TIME_Local1HzISR is a timer callback/PSP thing only and already exercised as part of a nominal run, not an application API so need to cover from a functional test point of view.  Just need to add error and nominal calls for CFE_TIME_Print to resolve this issue.

nmullane, 2021-08-05:
I'll work on this.

PR #1767 (2021-08-10): Fix #1692, Add misc time api functional test
**Describe the contribution**
Fixes #1692 

**Testing performed**
Steps taken to test the contribution:
1. Build and run cfe functional tests

**Expected behavior changes**
- No impact to behavior

**System(s) tested on**
 - Hardware: PC
 - OS: Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Niall Mullane - GSFC 582 Intern
commit e6979cf4 (2021-08-10): Fix #1692, Add misc time api functional test
commit f8ceda16 (2021-08-10): Merge pull request #1767 from nmullane/fix1692-misc-time-functional-tests

Fix #1692, Add misc time api functional test
PR #1772 (2021-08-11): cFE Integration candidate: 2021-08-10
**Describe the contribution**

PR #1737
- Fix #1733, Move global count into test global struct. 

PR #1722 
- Fix #806, Add ES application control API functional tests 
- Fix #1324, Add null pointer check to CFE_ES_ReloadApp functional test 

PR #1753 
- Fix #813, Add Generic Counter API test 

PR #1744 

- Fix #1743, update coverage test to use UtAssert macros 

PR #1734
  - Fix #1685
  - Fix #1688 
  - Fix #1686 
  - Fix #1687 

PR #1716 

- Fix #1645, Add resource ID API functional tests 

PR #1764 

- Fix #1763, last char truncated in coverage log outp [... cut]