nasa/cFE #1724: Scrub for discrepancies between implementation return codes and documentation

jphickey, 2021-07-28:
**Is your feature request related to a problem? Please describe.**
Need to perform a general scrub of API documentation for parameters, error checking, and and return codes, as was done for OSAL.

**Describe the solution you'd like**
- Confirm that all return/status codes which are _directly_ generated by an implementation appear in the documentation for that function as a `@retval` clause.
- Confirm that all return codes specified by a `@retval` clause also have a test case that covers them
- Confirm that parameter documentation is appropriately marked `@nonnull` or `@nonzero` where applicable
- Confirm that there is a test case that violates above rule to test function response w/bad input
- Confirm that coverage tests are not "overloaded" (i.e. violate only one rule at a time when confirming status code response).

**Additional context**
See nasa/osal#980

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-07-29:
Got the script running against CFE test logs, first set of results/issues for ES (unfortunately kind of a long list to look at):

```
OVERLOADED TEST: es_UT.c:3743 (09.047)
 --> matches CFE_ES_CreateChildTask[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> matches CFE_ES_CreateChildTask[NULL@1]=>CFE_ES_BAD_ARGUMENT
RETCODE: CFE_ES_AppID_ToIndex[CFE_ES_APPID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_AppID_ToIndex[NULL@1]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_CopyToCDS[CFE_ES_CDS_BAD_HANDLE@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_CopyToCDS[NULL@1]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_CounterID_ToIndex[CFE_ES_COUNTERID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_CounterID_ToIndex[NULL@1]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_CreateChildTask[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:3733 (09.045)
RETCODE: CFE_ES_CreateChildTask[NULL@1]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:3738 (09.046)
RETCODE: CFE_ES_CreateChildTask[NULL@2]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:3748 (09.048)
RETCODE: CFE_ES_CreateChildTask[ZERO@4]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_DeleteApp[CFE_ES_APPID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_DeleteChildTask[CFE_ES_TASKID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_DeleteGenCounter[CFE_ES_COUNTERID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_GetAppIDByName[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:3659 (09.032)
RETCODE: CFE_ES_GetAppIDByName[NULL@1]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:3664 (09.033)
RETCODE: CFE_ES_GetAppID[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:3655 (09.031)
RETCODE: CFE_ES_GetAppInfo[CFE_ES_APPID_UNDEFINED@1]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_GetAppInfo[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:1393 (04.069)
RETCODE: CFE_ES_GetAppName[CFE_ES_APPID_UNDEFINED@1]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_GetAppName[NULL@0]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_GetCDSBlockIDByName[NULL@0]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_GetCDSBlockIDByName[NULL@1]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4427 (11.071)
RETCODE: CFE_ES_GetCDSBlockName[CFE_ES_CDS_BAD_HANDLE@1]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_GetCDSBlockName[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4426 (11.070)
RETCODE: CFE_ES_GetGenCount[CFE_ES_COUNTERID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_GetGenCount[NULL@1]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4095 (10.024)
RETCODE: CFE_ES_GetGenCounterIDByName[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4101 (10.025)
RETCODE: CFE_ES_GetGenCounterIDByName[NULL@1]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4115 (10.033)
RETCODE: CFE_ES_GetGenCounterName[CFE_ES_COUNTERID_UNDEFINED@1]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_GetGenCounterName[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4114 (10.032)
RETCODE: CFE_ES_GetLibIDByName[NULL@0]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_GetLibIDByName[NULL@1]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:1845 (05.029)
[... 72 lines cut ...]
RETCODE: CFE_ES_RegisterCDS[ZERO@1]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_RegisterGenCounter[NULL@0]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4074 (10.019)
RETCODE: CFE_ES_RegisterGenCounter[NULL@1]=>CFE_ES_BAD_ARGUMENT
 --> validated by es_UT.c:4079 (10.020)
RETCODE: CFE_ES_ReloadApp[CFE_ES_APPID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_ReloadApp[NULL@1]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_RestartApp[CFE_ES_APPID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_RestoreFromCDS[CFE_ES_CDS_BAD_HANDLE@1]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_RestoreFromCDS[NULL@0]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_SetGenCount[CFE_ES_COUNTERID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_TaskID_ToIndex[CFE_ES_TASKID_UNDEFINED@0]=>CFE_ES_ERR_RESOURCEID_NOT_VALID
  NOT VALIDATED!!
RETCODE: CFE_ES_TaskID_ToIndex[NULL@1]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
RETCODE: CFE_ES_WriteToSysLog[NULL@0]=>CFE_ES_BAD_ARGUMENT
  NOT VALIDATED!!
```

astrogeco, 2021-08-19:
#1833 is in integration-candidate now

closed 2021-08-24 by skliper