nasa/cFE #1746: CFE_EVS_Register not null checking first

zanzaben, 2021-08-02:
**Describe the bug**
inside EVS register it clears the AppData record before it validates all the inputs. If some of the inputs are invalid that then causes the app to crash. 

**Expected behavior**
Validate inputs first before anything else. 

**Code snips**
https://github.com/nasa/cFE/blob/33a4f19ce6dc2c132823b3c8c487a59d5b4b81a8/modules/evs/fsw/src/cfe_evs.c#L46-L74

**Reporter Info**
Alex Campbell GSFC

skliper, 2021-08-02:
Could you provide a little more info on what actually causes a crash?  Was it observed or by analysis?  Both AppDataPtr and AppID are internal and success is checked, I'm not seeing how invalid parameters could cause a crash.

zanzaben, 2021-08-02:
So for example when FilterScheme != CFE_EVS_EventFilter_BINARY the function set the status to a failure code and stops. However  memset(AppDataPtr, 0, sizeof(EVS_AppData_t)); has already happened so it leaves the AppData as cleared out and never rebuilds it. This causes the functional test to print out this error "EVS Port1 66/1/CFE_EVS 41: App CFE_TEST_APP not registered with Event Services. Unable to send event." and then unable to finish the rests of the tests. 

I found this while trying to write the functional tests for this.

skliper, 2021-08-02:
> So for example when FilterScheme != CFE_EVS_EventFilter_BINARY the function set the status to a failure code and stops. However memset(AppDataPtr, 0, sizeof(EVS_AppData_t)); has already happened so [...]

Sounds to me like this is how it should work?  Either way (not calling `CFE_EVS_Register` or calling it with incorrect values) should leave the app in a state of not being able to send events.  You need to successfully register with EVS to be able to send events.  If it was working because AppDataPtr wasn't getting cleared that would be an actual bug.  Happy to chat if it helps.