nasa/cFE #1755: Improve consistency when working with OSAL status codes

jphickey, 2021-08-03:
**Is your feature request related to a problem? Please describe.**
Quite often in the CFE implementation, OSAL status codes are mixed with CFE status codes, often times sharing a single stack variable to hold either one, and even switching back and forth.

To improve clarity and consistency, OSAL status codes should be stored in a _separate_ stack variable from the CFE status codes.  Ideally, in the future, these should also migrate to a separate type (e.g. `osal_status_t`, as in nasa/osal#1108, or `CFE_Status_t`, as in nasa/cfe#921).  

**Describe the solution you'd like**
- Anywhere that an OSAL status code is stored on the stack, make a _separate_ variable.
- Always use a consistent name for that variable, e.g. `OsStatus`  (as opposed to "Status" or "ReturnCode" etc, which are vague).
- Avoid intermixing CFE status codes and OSAL status codes whenever possible (there will still be some times its necessary, when calling generic functions).  When it is necessary, make conversions between these types explicit/obvious via comments and/or type casting.
- When printing OSAL status codes in a log or event message, print it as long decimal (%ld), not as hex as the CFE status codes are.  This is because OSAL defines its status codes as decimal numbers, so logs/events should match this.

**Additional context**
This can be considered a prerequisite to nasa/cfe#921 (cannot introduce a unique `CFE_Status_t` type until it is no longer being used to store OSAL codes).

This would be a fix for next release (not 7.0.0).

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

zanzaben, 2021-08-03:
This is yet another addition to the growing list of problems due to OSAL error codes. #1676 #1672 #1723 #1675 #1599.

jphickey, 2021-08-03:
> This is yet another addition to the growing list of problems due to OSAL error codes. #1676 #1672 #1723 #1675 #1599.

Yes, I consider this related to #1676, as this addresses part of that one - however that previous issue was more general in nature.  This issue just deals with specifically having a separate variable for OSAL status vs. CFE status.

PR #1756 (2021-08-18): Fix #1755, separate variable for OSAL status
**Describe the contribution**
Ensure a separate stack variable, always named "OsStatus", is used to hold the result of an OSAL API call.  Do not use the same variable that is used to store a CFE status code.

This also fixes a few return code translations, where some functions were returning the OSAL status code (perhaps unintentionally) because it was stored in the same variable.

Fixes #1755

**Testing performed**
Build and sanity check CFE, run all tests

**Expected behavior changes**
Nothing significant, however some APIs which returned an OSAL  [... cut]
commit 42897cfc (2021-08-18): Fix #1755, separate variable for PSP status

Ensure a separate stack variable, always named "PspStatus", is used
to hold the result of a PSP API call.  Do not use the same variable
that is used to store a CFE status code.
commit 47eb39d5 (2021-08-18): Merge pull request #1756 from jphickey/fix-1755-osstatus-variable

Fix #1755, separate variable for OSAL status
commit b66f4767 (2021-08-18): Fix #1755, separate variable for OSAL status

Ensure a separate stack variable, always named "OsStatus", is used
to hold the result of an OSAL API call.  Do not use the same variable
that is used to store a CFE status code.
closed 2021-08-21 by astrogeco