nasa/cFE #1760: Write Resource ID functional test to check maximum

astrogeco, 2021-08-04:
[CCB:2021-08-04](https://github.com/nasa/cFS/wiki/Community-CCB-Agenda:-August-04-2021) APPROVED
- Open new also check that "ToIndex" is less than maximum in CounterID

_Originally posted by @astrogeco in https://github.com/nasa/cFE/pull/1734#issuecomment-892799313_

@nmullane and @jphickey, can you flesh this out ?

nmullane, 2021-08-05:
I'll work on this.

nmullane, 2021-08-05:
I can't edit the issue text, so should I create a new issue with a fleshed out description? Or the following can just be copied into this issue's description

---

**Is your feature request related to a problem? Please describe.**
Need open source functional tests for certifiability

**Describe the solution you'd like**
Add additional checks to functional tests for ES Resource ID to Index that ensure the returned index is less than the maximum allowed for that resource type

**Describe alternatives you've considered**
None

**Requester Info**
Niall Mullane - GSFC 582 Intern

jphickey, 2021-08-05:
I was able to update the description text.  I'll watch for the PR and review.  Thanks!

PR #1804 (2021-08-13): Fix #1760, check resource ID idx is less than max
**Describe the contribution**
Fixes #1760 

**Testing performed**
Steps taken to test the contribution:
1. Build and run cfe functional tests

**Expected behavior changes**
No impact to behavior

**System(s) tested on**
 - Hardware: PC
 - OS: Ubuntu 20.04

**Contributor Info - All information REQUIRED for consideration of pull request**
Niall Mullane - GSFC 582 Intern
commit 5fd8c21f (2021-08-13): Merge pull request #1804 from nmullane/fix1760-check-maximum-resource-id-index

Fix #1760, check resource ID idx is less than max
commit 6214ff27 (2021-08-13): Fix #1760, check resource ID idx is less than max

- Also deletes the created gen counter in TestCounterIDToIndex to clean
up
PR #1808 (2021-08-17): cFE Integration candidate 2021-08-17
**Describe the contribution**

PR #1790

- Fix #1784, add CFE assert macros to functional test

PR #1779 

Adds invalid id syslog to for CFE_ES_DeleteApp and CFE_ES_ReloadApp and verifies required reporting 

- Fix #1775 
- Fix #1773 

PR #1785 

Stop memory leak & add cds size test. 

- Fix #1776
- Fix #1770

PR #1765 
- Fix #1726, mark read only inputs as const 

PR #1804 

- Fix #1760, check resource ID idx is less than max 

PR #1801 

- Fix #1762, Update CFE_ES_RunLoop documentation 

PR #1745

- Fix #1677, Add Message Api Functional Test

PR #1798 
-  [... cut]
closed 2021-08-17 by astrogeco