nasa/cFE #1774: Replace cFE_FTAssert_VOIDCALL with new UtAssert_VOIDCALL

astrogeco, 2021-08-09:
> cFE_FTAssert_VOIDCALL from cfe_test.h can be replaced by the new UtAssert_VOIDCALL

Opening new issue for this

_Originally posted by @astrogeco in https://github.com/nasa/cFE/issues/1744#issuecomment-895286108_

@zanzaben can you flesh this issue out?

zanzaben, 2021-08-09:
The cFE Functional Test use a macro called cFE_FTAssert_VOIDCALL. That macro should be deleted and all existing references to it replaced with UtAssert_VOIDCALL.

jphickey, 2021-08-10:
In addition to this one I'd also like to replace these three as well (see #1790)

- `cFE_FTAssert_ResourceID_EQ` -> `CFE_UtAssert_RESOURCEID_EQ`
- `cFE_FTAssert_ResourceID_Undefined` -> `CFE_UtAssert_RESOURCEID_UNDEFINED`
- `cFE_FTAssert_NOT_CFE_SUCCESS` -> `CFE_UtAssert_NOT_SUCCESS`

Side note - We really shouldn't have generic macros in `cfe_testcase` app, this may not be the only "test" app that exists (users can add additional test apps for their apps/functions).  So CFE-specific assert macros should really be defined in the `cfe_assert` library, such that they can be used by user-defined test apps as well.

PR #1796 (2021-08-10): Fix #1774, replace VOIDCALL assert macro
**Describe the contribution**
Use the generic UtAssert_VOIDCALL macro now provided from UtAssert, rather than a local macro in cfe_testcase app.

Fixes #1774

**Testing performed**
Build and run all tests

**Expected behavior changes**
None (macros are equivalent)

**System(s) tested on**
Ubuntu

**Contributor Info - All information REQUIRED for consideration of pull request**
Joseph Hickey, Vantage Systems, Inc.
commit 25112ec7 (2021-08-10): Merge pull request #1796 from jphickey/fix-1774-replace-voidcall-macro

Fix #1774, replace VOIDCALL assert macro
commit 676c4731 (2021-08-10): Fix #1774, replace VOIDCALL assert macro

Use the generic UtAssert_VOIDCALL macro now provided from UtAssert,
rather than a local macro in cfe_testcase app.
PR #1772 (2021-08-11): cFE Integration candidate: 2021-08-10
**Describe the contribution**

PR #1737
- Fix #1733, Move global count into test global struct. 

PR #1722 
- Fix #806, Add ES application control API functional tests 
- Fix #1324, Add null pointer check to CFE_ES_ReloadApp functional test 

PR #1753 
- Fix #813, Add Generic Counter API test 

PR #1744 

- Fix #1743, update coverage test to use UtAssert macros 

PR #1734
  - Fix #1685
  - Fix #1688 
  - Fix #1686 
  - Fix #1687 

PR #1716 

- Fix #1645, Add resource ID API functional tests 

PR #1764 

- Fix #1763, last char truncated in coverage log outp [... cut]
closed 2021-08-11 by astrogeco