nasa/cFE #1799: CFE_SB_ReceiveBuffer does not return when the provided pipe is destroyed

jbohren-hbr, 2021-08-10:
**Describe the bug**
If a pipe is destroyed while `CFE_SB_ReceiveBuffer()` is waiting with `CFE_SB_PEND_FOREVER`, `CFE_SB_ReceiveBuffer()` either blocks indefinitely or returns `CFE_SUCCESS` (non-deterministically). This behavior is demonstrated in the proposed test case in PR #1777.

This is situation is only realizable in a multi-task context. The functional tests included in #1651 / #1657 do not test this behavior because they do not create multiple tasks in which one task is waiting on a blocking call to `CFE_SB_ReceiveBuffer()` while the pipe provided to it is deleted.

**To Reproduce**
See test case in PR #1777 

**Expected behavior**
The expectation (albeit not documented) is that if a pipe is deleted while waiting on `CFE_SB_ReceiveBuffer()`, that `CFE_SB_ReceiveBuffer()` would return immediately with an error code.

**Code snips**
See PR #1777 

**System observed on:**
See PR #1777 

**Additional context**
See PR #1777 

**Reporter Info**
Jonathan Bohren, Honeybee Robotics

skliper, 2021-09-07:
Worth noting I did just rediscover this caution about pipes not being inherently/originally designed as multi-thread safe:
https://github.com/nasa/cFE/blob/e5d4ed9a26ce3c991796494bc546341d5036b970/docs/cFE%20Application%20Developers%20Guide.md#L1536-L1541

jbohren-hbr, 2021-09-08:
> Worth noting I did just rediscover this caution about pipes not being inherently/originally designed as multi-thread safe:

Understood. Note that in this use case (implemented in SBN), only one thread is reading the pipe. However, a second thread is responsible for creating and destroying the pipe dynamically at runtime.