nasa/cFE #1817: Missing test cases for Event Services API parameter/return status combinations

jphickey, 2021-08-13:
**Is your feature request related to a problem? Please describe.**
The following parameter/return combinations are not exercised by the Event Services functional test:

```
 + condition: CFE_EVS_Register[any]=>CFE_ES_BAD_ARGUMENT
 + condition: CFE_EVS_Register[any]=>CFE_EVS_APP_FILTER_OVERLOAD
 + condition: CFE_EVS_Register[any]=>CFE_EVS_APP_ILLEGAL_APP_ID
 + condition: CFE_EVS_Register[any]=>CFE_EVS_UNKNOWN_FILTER
 + condition: CFE_EVS_Register[any]=>CFE_SUCCESS
 + condition: CFE_EVS_ResetAllFilters[any]=>CFE_EVS_APP_ILLEGAL_APP_ID
 + condition: CFE_EVS_ResetAllFilters[any]=>CFE_EVS_APP_NOT_REGISTERED
 + condition: CFE_EVS_ResetFilter[any]=>CFE_EVS_APP_ILLEGAL_APP_ID
 + condition: CFE_EVS_ResetFilter[any]=>CFE_EVS_APP_NOT_REGISTERED
 + condition: CFE_EVS_SendEvent[any]=>CFE_EVS_APP_ILLEGAL_APP_ID
 + condition: CFE_EVS_SendEvent[any]=>CFE_EVS_APP_NOT_REGISTERED
 + condition: CFE_EVS_SendEventWithAppID[any]=>CFE_EVS_APP_NOT_REGISTERED
 + condition: CFE_EVS_SendTimedEvent[any]=>CFE_EVS_APP_ILLEGAL_APP_ID
 + condition: CFE_EVS_SendTimedEvent[any]=>CFE_EVS_APP_NOT_REGISTERED
```

**Describe the solution you'd like**
Ideally should be tested, but this may be difficult with the current cfe_testcase app because it uses event services to report test results.

**Describe alternatives you've considered**
Document why these cannot be tested in the current Caelum release, and rely on coverage testing to test these responses (which is already implemented)

**Additional context**
Part of/related to #1724 (each subsystem will be handled separately)

These should generally not be marked as `covtest` in the doxygen because they (theoretically) _can_ be tested in a functional test, its just that the cfe_testcase app uses and registers with event services before starting tests, so any interference with this would prevent/conflict with normal test reporting.  This is true for all calls to `CFE_EVS_Register` (because its already registered), and all items that return  `CFE_EVS_APP_ILLEGAL_APP_ID` or `CFE_EVS_APP_NOT_REGISTERED`  (which is really everything in the above list).  These would all need a _separate_ app/task that is not registered with EVS to accomplish.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2021-08-13:
Agree 100%, the registration success "test case" is by demonstration/use and all the errors will be traced to "analysis" which translates to coverage for now.  Improvements can be implemented in the next round w/ additional test apps.  Added the "functional" label to specifically track this sort of issue (and "unit-test" still generically identifies the issue as not applying to "production" code)

jphickey, 2021-08-23:
Dependent on #1761 - all these EVS cases would require a companion/secondary test app that is _not_ registered with EVS, in order to exercise the various EVS registration-related calls.