nasa/cFE #1827: SB coverage test - need to verify MsgSendErrorCounter increments when msg too big

skliper, 2021-08-16:
**Is your feature request related to a problem? Please describe.**
Missing verification of MsgSendErrorCounter increment on message too big (requirement)

**Describe the solution you'd like**
Add verification

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-08-17:
I wanted to do this, but we do not expose the counter other than in the housekeeping telemetry stream.  So without subscribing to the TLM packet somehow, we can't read the counter (unless there was something I missed?).  I assume a CMD/TLM-based test could do this more easily.

jphickey, 2021-08-17:
I suppose the test could subscribe to the SB HK TLM message, but then you'd have a dependency on SCH running, unless we also generated the SEND_HK command locally too?

PR #1856 (2021-08-20): Fix #1827 Check error ctr to TransmitMsg test
**Describe the contribution**
Fixes #1827 Verify error counter increments when using TransmitMsg for a message that is too big.

**Testing performed**
Steps taken to test the contribution:
1. `make SIMULATION=native ENABLE_UNIT_TESTS=true OMIT_DEPRECATED=BUILDTYPE=release prep`
2. `make -C build/native/default_cpu1/sb`
3. `make -C build/native/default_cpu1/sb test`
4. `make`
5. `make test`

**Expected behavior changes**
None.

**System(s) tested on**
 - OS: Ubuntu 18.04 VM

**Contributor Info - All information REQUIRED for consideration of pull req [... cut]
commit 748b6f75 (2021-08-20): Fix #1827 Check error ctr to TransmitMsg test
commit e4150b0c (2021-08-20): Merge pull request #1856 from pepepr08/fix1827-sb-errorctr

Fix #1827 Check error ctr to TransmitMsg test
closed 2021-08-21 by astrogeco