nasa/cFE #1829: SB functional test - exercise CFE_SB_ReceiveBuffer CFE_SB_PEND_FOREVER and CFE_SB_POLL options

skliper, 2021-08-17:
**Is your feature request related to a problem? Please describe.**
SB functional test only calls CFE_SB_ReceiveBuffer with a timeout of 100, requirements include poll and pend forever.

**Describe the solution you'd like**
Exercise CFE_SB_POLL and CFE_SB_PEND_FOREVER options.

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-08-17:
I actually added a couple calls to ReceiveBuffer w/CFE_SB_POLL as part of #1812.

I can add a call w/CFE_SB_PEND_FOREVER - didn't do this at first because if the queue is empty, that would hang the test, but that would also be a failure so probably ok?  Obviously won't be able to test the case of PEND_FOREVER + empty queue....

skliper, 2021-08-17:
Great!  Please add a PEND_FOREVER call with a message on the bus so it returns immediately, no need for a test that actually pends forever.  It'd also be good for the final test where it does pend for a short timeout to actually confirm at least a minimum amount of time passed.

PR #1885 (2021-09-01): cFE Integration candidate: 2021-08-31
**Describe the contribution**

PR #1870 

- Fix #1812, Add SB API test cases

PR #1869 

- Fix #1866, Add ES API test cases

PR #1872 

- Fix #1868, Add TBL API test cases

PR #1871 

- Fix #1867, Add FS API test cases

PR #1860 

- Fix #1843, Add Time Clock Test 

PR #1862 

EVS Coverage
- Fixes #1819, Adds test for MessageSendCounter not incrementing when sending a filtered event.
- Fixes #1820, Adds tests for LogOverflowCounter incrementing when the log overflows.

PR #1876

SB test improvements
- Fix #1825 
- Fix #1826
- Fix #1829
- Fix #1830

    
PR  [... cut]
commit 5c388e31 (2021-09-01): Fix #1829, confirm CFE_SB_PEND_FOREVER flag

Use the CFE_SB_PEND_FOREVER flag on a few of the calls to CFE_SB_ReceiveBuffer
to confirm that this works as expected.  Previously only CFE_SB_POLL and
a nonzero timeout were used.

Note this can only be used when it is known/expected that the message
queue is not empty.  If there is any possiblility that the queue is empty,
then this cannot be used or else the test may block.  This also means that
if the test fails, it may fail by blocking and never finishing the test
rather than an actual FAIL testcase.
closed 2021-09-01 by astrogeco