nasa/cFE #1882: FS missing branch coverage of run-time endian logic (and sometimes line)

skliper, 2021-08-25:
**Is your feature request related to a problem? Please describe.**
FS suffers from the unfortunate pattern of run time checks for endian (and associated logic), which leads to uncovered branches and possibly uncovered code (depending on endian of platform).

**Describe the solution you'd like**
Complete coverage is possible if run on each system, but should be resolved in the future.  Possibly in concert with #1209 and related abstraction such that this code doesn't care (whatever underlying API/implementation that handles raw data should do the appropriate conversion).

**Describe alternatives you've considered**
None

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

avan989, 2023-06-02:
"EndianCheck" is hard-coded to 0x01020304. Not possible for a complete code coverage without changing the implementation. Same for CFE_FS_WriteHeade, CFE_FS_ReadHeader, CFE_FS_SetTimestamp. Recommend closing ticket as the branch cannot be reached. @dmknutsen @dzbaker

skliper, 2023-06-02:
@avan989 @dmknutsen @dzbaker - these tickets were meant to track the fact we can't get full coverage w/ the current setup.  I recommend against closing them unless the lack of coverage is addressed.