nasa/cFE #1919: SB missing coverage in CFE_SB_AppInit, alternate config

skliper, 2021-08-31:
**Is your feature request related to a problem? Please describe.**
Uncovered code since it would require an alternate config where CFE_PLATFORM_EVS_MAX_EVENT_FILTERS is less than the defined number of filters:
```
     190                 :            :     /* Be sure the number of events to register for filtering
     191                 :            :     ** does not exceed CFE_PLATFORM_EVS_MAX_EVENT_FILTERS */
     192         [ -  + ]:         12 :     if (CFE_PLATFORM_EVS_MAX_EVENT_FILTERS < CfgFileEventsToFilter)
     193                 :            :     {
     194                 :          0 :         CfgFileEventsToFilter = CFE_PLATFORM_EVS_MAX_EVENT_FILTERS;
     195                 :            :     
```
here:
https://github.com/nasa/cFE/blob/5e41330979586bcdb113f491ea6c38ff9975fc72/modules/sb/fsw/src/cfe_sb_task.c#L190-L195

**Describe the solution you'd like**
Consider refactor of self-filtering, unfriendly to ground management of filters.  Generated events shouldn't spam to begin with, and be useful vs requiring self-filtering.  Or if this pattern is retained the silent misconfiguration isn't great, could better utilize EVS registration error reporting.  Either way this design isn't great.  Could get coverage by modifying config, but really it'd just be broken (have user fix config vs silently not doing what the user requested).

**Describe alternatives you've considered**
See above.

**Additional context**
None

**Requester Info**
Jacob Hageman - NASA/GSFC

jphickey, 2021-08-31:
See also #1669 .... I had also observed this when writing the functional test (can close the other as dupe).

avan989, 2023-06-09:
Code Coverage is not possible without modifying the default configuration.
Currently:
1. CFE_PLATFORM_EVS_MAX_EVENT_FILTERS is config to 8 
2. CfgFileEventsToFilter is hardcoded to 0.

Reference ticket for the recommended fix.