nasa/cFE #1932: ES missing coverage in CFE_ES_PoolCreateEx, requires alternate config

skliper, 2021-09-05:
**Is your feature request related to a problem? Please describe.**
Branch/line below uncovered since it would require an alternate configuration:
```
     284                 :         11 :     Alignment = ALIGN_OF(CFE_ES_PoolAlign_t); /* memory mapped pools should be aligned */
     285         [ -  + ]:         11 :     if (Alignment < CFE_PLATFORM_ES_MEMPOOL_ALIGN_SIZE_MIN)
     286                 :            :     {
     287                 :            :         /*
     288                 :            :          * Note about path coverage testing - depending on the
     289                 :            :          * system architecture and configuration this line may be
     290                 :            :          * unreachable.  This is OK.
     291                 :            :          */
     292                 :          0 :         Alignment = CFE_PLATFORM_ES_MEMPOOL_ALIGN_SIZE_MIN;
     293                 :            :     }
```
here:
https://github.com/nasa/cFE/blob/e5d4ed9a26ce3c991796494bc546341d5036b970/modules/es/fsw/src/cfe_es_mempool.c#L284-L293

**Describe the solution you'd like**
Test with alternate configurations

**Describe alternatives you've considered**
None

**Additional context**
Jacob Hageman - NASA/GSFC

**Requester Info**
Full name and company/organization if applicable

avan989, 2023-06-12:
code coverage is not possible without changing the configuration. 

Currently: Ran on docker with ubuntu 22.10 on windows
1. Alignment = 16
2. CFE_PLATFORM_ES_MEMPOOL_ALIGN_SIZE_MIN  = 4

jphickey, 2023-06-12:
You could add the minimum alignment as a passed-in parameter - then this new function could be called from unit test with a larger value.  However because `CFE_ES_PoolCreateEx()` is part of the API, you'd need to rename the existing function to add the param, then make a compatible CFE_ES_PoolCreateEx() that calls the new function with CFE_PLATFORM_ES_MEMPOOL_ALIGN_SIZE_MIN.