nasa/cFE #1937: Missing entries in system log during initialization

dmknutsen, 2021-09-07:
**Describe the bug**
The first ~5 entries during initialization are being dropped from the system log.  

**To Reproduce**
Dump the system log and review the first few entries.  

See the attached cfs_cfs_stdout.txt file and compare against a what was captured in the system log - SysLog.txt.   Specifically, the message to determine power on/processor reset is being dropped - which is sometimes used for requirement verification/is resulting in test failures.

**Expected behavior**
All log entries captured in log.

**Code snips**
If applicable, add references to the software.

**System observed on:**
Oracle VM VirtualBox
OS: ubuntu-20.10
cFS Versions: cfe v6.8.0-rc1+dev593, osal v5.1.0-rc1+dev458, psp v1.5.0-rc1+dev112.

**Reporter Info**
Dan Knutsen
NASA/Goddard

[cfs_cfs_stdout.txt](https://github.com/nasa/cFE/files/7121484/cfs_cfs_stdout.txt)
[SysLog.txt](https://github.com/nasa/cFE/files/7121486/SysLog.txt)

skliper, 2021-09-07:
Wasn't able to recreate w/ open source
1. Start cFS
2. Command a dump of the syslog (FC=11, APID=6)
```
./cmdUtil -ELE -C11 -A6 -s64:/cf/syslog.txt 
```
3. View the file in the local cf directory... all entries exist.

dmknutsen, 2021-09-07:
[SysLogRaw.txt](https://github.com/nasa/cFE/files/7122779/SysLogRaw.txt)

skliper, 2021-09-07:
Turns out the log was just getting overwritten.  Increased the log size (increased the distribution definition of CFE_PLATFORM_ES_SYSTEM_LOG_SIZE to 13072) and the initial messages were observed.  Closing as invalid.

closed 2021-09-07 by skliper