nasa/cFE #1952: Potential for recursive loop of Event TLM MsgId is incorrect

jphickey, 2021-09-16:
**Describe the bug**
If the software bus `CFE_SB_TransmitMsg()` fails to send a message due to a validation failure, it will send an event through EVS.

Event Services, in turn, generates a message (LongEventTlm/ShortEventTlm) which is broadcast via software bus.

Howevver if the Event Telemetry MID value (`CFE_EVS_LONG_EVENT_MSG_MID`) is not set correctly (or some other EVS config is bad) such that EVS tries to send event messages which do not validate, a recursive loop ensues and the software eventually segfaults.

**To Reproduce**
(Mis)configure `CFE_EVS_LONG_EVENT_MSG_MID` to a value which will not pass the CFE_SB_TransmitMsgValidate tests.  Run CFE, it will get in a recursive loop and eventually segfault/crash as soon as any app sends an event.

**Expected behavior**
Should not do a recursive loop

**System observed on:**
Ubuntu

**Additional context**
Should be protection against recursive event loops like this, where if an event fails to send, it should not cause another event to be sent.   This protection appears this is not working correctly right now, at least not for EVS messages.

__NOTE__: To be absolutely clear - the issue described here is a mis-configuration issue.  It will not happen in a properly configured system, so long as EVS generates messages which are "transmittable".  BUT - there are other events that might be triggered by a CFE_SB_TransmitMsg call, such as a MsgLim error, and its not clear of a similar recursive loop might be possible there (have not tested/investigated).

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2021-09-16:
It seems the "recursive event" protection is only enabled for CFE_SB_SEND_NO_SUBS_EID.  Not sure if this is a historical thing (i.e. if there is a reason this was only done for that particular event) or something recently introduced/oversight with the caelum changes.

See Here:
https://github.com/nasa/cFE/blob/64a6a59456fa9e47dc93e4bb9cecacc3d86d1862/modules/sb/fsw/src/cfe_sb_api.c#L1527-L1537

However the same "RequestToSendEvent" is not done for the other events that this function might generate.

For a short term fix, any+all events that might be generated from the `CFE_SB_TransmitMsg` path should be wrapped with this same check.

However, from an architectural standpoint, it doesn't make a whole lot of sense for this recursion protection to be in SB at all.  It's really an EVS problem, and SB seems to be protecting EVS from itself here.

Rather, EVS should track its per-task state, and if a task tries to send the same event ID while in the process of sending that event ID, it should drop it itself, to break the loop.

jphickey, 2021-09-16:
A skim through the other functions in this path (CFE_SB_TransmitMsg, CFE_SB_BroadcastBufferToRoute) show that other events also _do_ have this protection:

- CFE_SB_GET_BUF_ERR_EID
- CFE_SB_MSGID_LIM_ERR_EID
- CFE_SB_Q_FULL_ERR_EID
- CFE_SB_Q_WR_ERR_EID

So the problem is likely limited to these Event IDs specifically from `CFE_SB_TransmitMsgValidate`:

- CFE_SB_SEND_BAD_ARG_EID
- CFE_SB_SEND_INV_MSGID_EID
- CFE_SB_MSG_TOO_BIG_EID

All of these EIDs point to a configuration issue as opposed to a runtime issue, which reduces the severity of this issue.

jphickey, 2021-09-16:
But important to note in a longer term/system level viewpoint, since Caelum allows components to be overridden/modified, it is not really appropriate for SB to "know" that EVS calls `CFE_SB_TransmitMsg()` as part of its work and therefore specifically protect those EIDs from recursion.   For example, EVS also calls `CFE_MSG_Init()` and `CFE_ES_GetAppName()` (just to name a couple).  If any of these generate an event, the same issue exists.

It really calls for EVS to protect itself from recursion in a more generic sense, to ensure that ANY event from ANY app will not get itself into a loop and crash the system.

skliper, 2021-09-19:
I'm in favor of treating the current PR as an optional patch if folks want it (since it's a misconfiguration issue), but plan for protecting from within EVS in a future release.

PR #2001 (2021-12-06): cFE Integration Candidate, Caelum+dev1
**Describe the contribution**

PR #1663
- Fix #1546, use cfe time clock state enum t in cmd/tlm and handling

PR #1974
- Fix #1945, add CFE_SB_ValueToMsgId/MsgIdToValue wrappers
- Fix #1944, use 0 as the invalid CFE_SB_MsgId_t value
- Fix #245, use type safe value for CFE_SB_MsgId_t 

PR #1973 
- Fix #1818, EVS functional test add for different event types

PR #1954
- Fix #1952, Patch for recursive event loop
 
PR #1911
- Fix #1300, CFE_ES_MemAddOff structure to associated CFE_ES_MemAddress_t and CFE_ES_MemOffset_t

PR #1986
- Fix #1984, Avoid aliasing wa [... cut]
PR #1954 (2021-12-06): Fix #1952, patch for recursive event loop
**Describe the contribution**
Adds CFE_SB_RequestToSendEvent/CFE_SB_FinishSendEvent wrappers around all events generated by CFE_SB_TransmitMsgValidate.  This is avoids the potential for a recursive loop if configured improperly.

Fixes #1952

**Testing performed**
Build and sanity check CFE, run all tests.

**Expected behavior changes**
Tested with the mis-configuration where EVS had an invalid MID and confirmed that the system does not get into a recursive loop and segfault anymore.

**System(s) tested on**
Ubuntu

**Additional context**
This is just  [... cut]
commit 92cc55e5 (2021-12-06): Fix #1952, patch for recursive event loop

Adds CFE_SB_RequestToSendEvent/CFE_SB_FinishSendEvent wrappers around
all events generated by CFE_SB_TransmitMsgValidate.  This is avoids
the potential for a recursive loop if configured improperly.
commit dd62c2e2 (2021-12-06): Merge pull request #1954 from jphickey/fix-1952-recursive-event

Fix #1952, patch for recursive event loop
closed 2021-12-06 by astrogeco