nasa/cFE #1985: Check Return Value of CFE_ES_PutPoolBuf

ArielSAdamsNASA, 2021-10-01:
**Describe the bug**
Calling CFE_ES_PutPoolBuf without checking return value (as is done elsewhere 4 out of 5 times).

**Expected behavior**
Check return value of CFE_ES_PutPoolBuf as seen elsewhere. 

**Code snips**
https://github.com/nasa/cFE/blob/98f78e8604c19415fd1e199eae94196a781539b8/modules/sb/fsw/src/cfe_sb_buf.c#L167

**System observed on:**
Coverity: https://scan.coverity.com/projects/arielsadamsnasa-cfs-jsf-rules?tab=overview

**Additional context**
Examples of return value being checked for CFE_ES_PutPoolBuf

https://github.com/nasa/cFE/blob/98f78e8604c19415fd1e199eae94196a781539b8/modules/sb/fsw/src/cfe_sb_buf.c#L258-L259

https://github.com/nasa/cFE/blob/5e41330979586bcdb113f491ea6c38ff9975fc72/modules/tbl/fsw/src/cfe_tbl_internal.c#L354-L362

**Reporter Info**
Ariel Adams, ASRC Federal

thnkslprpt, 2022-12-29:
I put in  PR just adding a WriteToSysLog.

Layomx, 2026-11-01:
HI, I was doing a review about this issue, I saw the current implementation and the previous PR. I also read the discussion. Is this issue still available so I can work on a possible fix?