nasa/cFE #2104: Link to "cFE Users Guide" in the top-level README 404s.

acarrillo, 2022-05-17:
**Describe the bug**
This is the URL to the `cFE_Users_Guide.pdf` given on the front page README.md: https://github.com/nasa/cFS/blob/gh-pages/cFE_Users_Guide.pdf

The URL returns a 404.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/nasa/cFE/blob/main/README.md
2. Follow the link given by:
> The detailed cFE user's guide can be viewed at https://github.com/nasa/cFS/blob/gh-pages/cFE_Users_Guide.pdf.

3. Click the link and see the 404.

**Expected behavior**
The document should exist at the given URL.

**Reporter Info**
Alex Carrillo

skliper, 2022-05-19:
Correct link is now https://github.com/nasa/cFE/blob/gh-pages/cfe-usersguide.pdf

acarrillo, 2022-05-24:
Thank you @skliper !

PR #2111 (2022-05-31): cFE Integration candidate: Caelum-rc4+dev10
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFE/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**

PR #2105

- Fix #2100, Resolve UT uninit var static analysis warnings

- Fix #2101, Resolve fsw uninit var static analysis warnings

- Fi [... cut]
PR #2108 (2022-05-31): Fix #2104, cFE_Users_guide.pdf 404
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFE/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
A clear and concise description of what the contribution is.
- Fixes #2104 

**Expected behavior changes**
A clear and concise description of how t [... cut]
closed 2022-05-31 by astrogeco