nasa/cFE #2112: Commit and format check doesn't execute when PR title format fails

astrogeco, 2022-05-31:
**Is your feature request related to a problem? Please describe.**
When running against a pull request, if the PRs title fails the format check, the workflow skips the commit message check. 

**Describe the solution you'd like**
Workflow checks both PR and commit titles regardless of failures

**Describe alternatives you've considered**
- Separate into parallel jobs. 
- Use a "keep going on failure" flag for the step (not sure if this is an option)

**Additional context**
See https://github.com/nasa/cFE/actions/runs/2414940003

**Requester Info**
Gerardo E. Cruz-Ortiz, NASA

passionInfinite, 2022-08-04:
Hello @astrogeco ,
I am happy to take this one. Let me know if this is still available to work upon. Thanks!

dzbaker, 2022-08-05:
> Hello @astrogeco , I am happy to take this one. Let me know if this is still available to work upon. Thanks!

Thanks @passionInfinite! it looks like this one was addressed by nasa/cFS#516. Let us know, though, if there is another issue you would be interested in taking on.