nasa/cFE #2140: RunApp table scan waits maximum time for a shutdown app.

excaliburtb, 2022-08-29:
**Describe the bug**
If ES is commanded to stop an app, the RunAppTableScan background task doesn't check that the app has reached the desired state of CFE_ES_AppState_STOPPED. Instead it continues to decrement the wait time, which means the app will never get cleaned up until the timespan of `CFE_PLATFORM_ES_APP_KILL_TIMEOUT * CFE_PLATFORM_ES_APP_SCAN_RATE` is reached.

**To Reproduce**
Steps to reproduce the behavior:
1. Set CFE_PLATFORM_ES_APP_SCAN_RATE  to 1000, and set CFE_PLATFORM_ES_APP_KILL_TIMEOUT  large enough to visually inspect the time between an app reaching CFE_ES_ExitApp and CFE sending the event with `Stop Application <AppName> Completed`
2. Start CFS, command ES to stop the target app. Note that the Stop Application takes the full CFE_PLATFORM_ES_APP_KILL_TIMEOUT  before the Stop Application event is generated.

**Expected behavior**
I think the intent of the RunAppTableScan is to detect when an app is immediately able to be cleaned up or a timeout has been reached. The timeout is in place but the CFE_ES_AppState_STOPPED condition isn't checked.

**Code snips**
If applicable, add references to the software.
Replacing the 
`            if (AppPtr->AppState > CFE_ES_AppState_RUNNING)`
code block in `CFE_ES_RunAppTableScan`
with something like:
```
            if (AppPtr->AppState > CFE_ES_AppState_RUNNING)
            {   
                /*
                 * Increment the "pending" counter which reflects
                 * the number of apps that are in some phase of clean up.
                 */
                ++State->PendingAppStateChanges;
                
                /*
                 * Decrement the wait timer, if active.
                 * When the timeout value becomes zero, take the action to delete/restart/reload the app
                 */
                if (AppPtr->AppState == CFE_ES_AppState_STOPPED || AppPtr->ControlReq.AppTimerMsec <= ElapsedTime)
                {
                    AppPtr->ControlReq.AppTimerMsec = 0;

                    /* Add it to the list to be processed later */
                    AppTimeoutList[NumAppTimeouts] = CFE_ES_AppRecordGetID(AppPtr);
                    ++NumAppTimeouts;
                }
                else
                {   
                    AppPtr->ControlReq.AppTimerMsec -= ElapsedTime;
                }
            }
```
improves this app unload delay.

**System observed on:**
 - Hardware
 - OS: [e.g. Linux 4.4]
 - Versions [e.g. cFE 6.6, OSAL 4.2, PSP 1.3 for mcp750, any related apps]
 Anything

**Additional context**
Add any other context about the problem here.

**Reporter Info**
Full name and company/organization if applicable
Thomas Brain - METECS contractor to JSC.

dmknutsen, 2022-08-29:
@jphickey - Does the above suggestion match the design intent of the app table scan loop?

jphickey, 2022-09-13:
Just saw this mention (sorry for delay!) ... I think there was some historical precedent for always implementing the time delay on resource recovery/cleanup even if the application did voluntarily call `CFE_ES_ExitApp()`.  However with the current code I don't see any real necessity for that delay.  The suggested change of adding a check for the STOPPED state to accelerate the process of recovering the resources seems reasonable to me.